stephane/libmodbus #246: Is a libmodbus context really thread-safe?

mhei, 2015-02-20:
Documentation states:
> A libmodbus 'context' is thread safe and may be shared among as many application
> threads as necessary, without any additional locking required on the part of the caller.

I wonder whether this is really the case (any longer). In `struct _modbus` there are fields which are `struct timeval` for example, which consists of two sub-fields which are updated in two steps. And both sub-fields are non-atomic types, thought one can assume that updating one field is atomic, can't we?

I also wonder whether is is really necessary or important, that libmodbus is thread-safe? Modbus itself is a master-slave protocol, so there should be no parallelism. Maybe anybody could illustrate an usage scenario where multiple threads are involved?

fengyikil, 2015-06-04:
I tested it ,and found many "ERROR Connection timed out: " , it seems not thread-safe.

stephane, 2015-06-09:
Thank you for the report

closed 2015-06-09 by stephane

bigzwey, 2018-01-19:
I find it useful to create multiple servers simultaneously on one machine, each on a different port and using different logic and/or registers.   It seems possible that some of the function calls, pointed to by the context, could be interrupted partway through.

pboettch, 2018-01-19:
@bigzwey Interrupted by what? Libmodbus is a library and does not contain any shared-between-threads-data in its context(s). (If it does it would be a bug).

If you use libmodbus in your threaded code it is up to you to take care of the thread-safety by using mutexes when accessing shared-memory (for example the `modbus_ctx`).

commit b7247f10 (2015-06-09): A libmodbus context isn't thread safe (closes #246)