stephane/libmodbus #248: Installing libmodbus using Cygwin

xristosx, 2015-03-04:
I am trying to install libmodbus on a Windows 8.1 machine, using Cygwin.
and I am getting the following error when I type "make":
configure: WARNING: winsock2.h: present but cannot be compiled
configure: WARNING: winsock2.h:     check for missing prerequisite headers?
configure: WARNING: winsock2.h: see the Autoconf documentation
configure: WARNING: winsock2.h:     section "Present But Cannot Be Compiled"
configure: WARNING: winsock2.h: proceeding with the compiler's result
configure: WARNING:     ## -----------------------------------------------------                                    ------ ##
configure: WARNING:     ## Report this to https://github.com/stephane/libmodbus/                                    issues ##
configure: WARNING:     ## -----------------------------------------------------                                    ------ ##
checking for winsock2.h... no
checking whether TIOCSRS485 is declared... no
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating src/Makefile
config.status: creating src/modbus-version.h
config.status: creating doc/Makefile
config.status: creating tests/Makefile
config.status: creating libmodbus.pc
config.status: creating libmodbus.spec
config.status: creating config.h
config.status: creating tests/unit-test.h
config.status: executing depfiles commands
config.status: executing libtool commands

Any ideas?

Thank you in advance.

dwilliamsvancouv, 2015-06-17:
I get the exact same result when I try to install libmodbus 3.1.2 on windows 7 using Cygwin. I don't need to use modbus over tcp. I am hoping this problem with winsock2 will not affect me. Am I right in assuming this?

stephane, 2015-08-04:
Sorry but I can't help you on Windows support. I've already do my best to support libmodbus on Windows XP but this OS is just a gaming platform for me.

Are you able to provide a patch?

Ketorin, 2015-08-24:
I got 3.0.6 to compile by simply commenting out winsock 2, I think.

stephane, 2016-09-09:
Could we close this issue?

R030t1, 2017-05-02:
I'm going to be adding some bugs related to Windows support soon. If you don't want to support Windows you should really drop it from the project's description. It seems like Windows support was and has continued to be absent.

stephane, 2017-05-02:
If you have a look to the source code, you'll see I included many tricks to support Windows and I run the unit tests against Windows at each new release but I don't use Windows for work or production so Windows updates or changes can break things.

BTW libmodbus v3.0 doesn't receive changes anymore (see v3.1.x and up).

stephane, 2017-05-02:
winsock 2 must be installed for TCP support.

closed 2017-05-02 by stephane

R030t1, 2017-05-02:
Just because you have Windows-specific code doesn't mean the library builds and runs successfully on Windows.

stephane, 2017-05-05:
@R030t1 That's why I run the unit tests on Windows myself (as said in my previous comment).

@oldfaber and @StalderT have provided valuable patches to libmodbus for Windows support but if the users of libmodbus under Windows just complain w/o providing patches I'll just remove the Windows support.

BTW Your comment wasn't helpful to improve the project.

R030t1, 2017-05-05:
I realize it's not the most helpful of comments but this is one of many projects I have seen that has a POSIX or GNU/Linux codebase that is very hard to build on Windows. I admit I don't know why for every issue, but a lot of the errors seem like basic things that any reasonable amount of testing should catch.

Just because it is possible to build it on your installation of Windows does not mean other people can build it, and some of the comments I have seen dismiss what are likely valid build issues. If you don't have the time to create a clean installation of Windows (in e.g. a virtual machine) to test from the ground up every once and a while then I'm not sure you have the time necessary to support Windows.

For this issue specifically, the error seems to be saying that WinSock2 is available but something else is wrong. There is no way he does not have WinSock2 installed - it comes with all recent versions of the OS. You've closed this bug without understanding why what is going on is happening.