stephane/libmodbus #252: OS X build: HAVE_STRLCPY doesn't work

terman110, 2015-03-16:
Hi guys,

I tried to compile on OS X libmodbus on OS X 10.9 and it failed due to strlcpy is already defined. The following small pathes resolved the issue. 

libmodbus/libmodbus-3.1.2/src/modbus.c

    1877      free(mb_mapping);
    1878   }
    1879   
    1880  -#ifndef HAVE_STRLCPY
    1880  +#if (!defined __APPLE__) && (!defined HAVE_STRLCPY)
    1881   /*
    1882    * Function strlcpy was originally developed by
    1883    * Todd C. Miller <Todd.Miller@courtesan.com> to simplify writing secure code.

libmodbus/libmodbus-3.1.2/src/modbus-private.h

    118   void _error_print(modbus_t *ctx, const char *context);
    119   int _modbus_receive_msg(modbus_t *ctx, uint8_t *msg, msg_type_t msg_type);
    120 
    121   -#ifndef HAVE_STRLCPY
    121   +#if (defined __APPLE__)
    121   +    #include <string.h>
    121   +#elif (!defined HAVE_STRLCPY)
    122    size_t strlcpy(char *dest, const char *src, size_t dest_size);
    122   #endif

Please consider this fix. Thanks a lot.

stephane, 2015-05-21:
I built libmodbus on OS 10.9 and 10.10 but I didn't see this problem.
I seems to be a workaround for a bug in autotools?
Which versions do you use?

stephane, 2015-08-04:
@terman110 any news on this front?

stephane, 2015-08-27:
I hope it was just a false positive and the issue was elsewhere.
Feel free to reopen if necessary.

closed 2015-08-27 by stephane