stephane/libmodbus #253: compile error

rpiModbus, 2015-03-25:
hi, i'm trying to make the rs485 work with libmodbus, i've followed all the topic, applied the patch but with no results, now, when make i have the following error:

../src/.libs/libmodbus.so: undefined reference to `digitalWrite'
../src/.libs/libmodbus.so: undefined reference to `pinMode'
../src/.libs/libmodbus.so: undefined reference to `wiringPiSetup'
../src/.libs/libmodbus.so: undefined reference to `pullUpDnControl'

it seems that libmodbus doesn't find wiringPi library, someone could help me?

thanks in advance

dhruvvyas90, 2015-03-25:
Hello,

You will have to add lwiringpi while configuring libmodbus. (after running ./augogen.sh. Use ./configure --help to see what all parameters you can set. As far as I remember you need to set LIBS=-lwiringpi, not sure though). Also you will include wiringpi.h file in source (.c) file to make use of wiringpi functions in the file.

Also, it's advisable to use google group page of libmodbus to ask such question(s) since it's not a libmodbus issue.

Hope it helps.

regards,

Dhruv

rpiModbus, 2015-03-26:
hi, tahnks for the answer, i solved with your help. now i have another problem:
i have this board with max485 https://www.homegear.eu/images/1/10/RS485_Serial_Module_Wiring.png and have connected with DE,RE signals together. now my problem is that when i start the transmission I have the right output signal, but when i receive i have nothing, as you can see in the image in the attachment. 

i've replaced the max485 with a texas instruments ic http://www.ti.com/lit/ds/slls877f/slls877f.pdf , they are pin compatible, but this one is suited for 3.3V applications, but nothing seems to happen.
i use an SDM630 multimeter to talk with modbus, in the idle times it puts on the bus a generic signal at a very high frequency, is it normal?
thanks in advance an sorry for my english. for further infos please ask me.
[image]

 P.S.
as you suggested i've written on the libmodbus group, but still have no answer, https://groups.google.com/forum/#!forum/libmodbus

sorry for crossposting

Chadster766, 2015-03-26:
I'm curious how you captured the serial coms into a nice chart like that.

rpiModbus, 2015-03-26:
picoscope....

Chadster766, 2015-03-26:
Interesting, what model do you recommend?

dhruvvyas90, 2015-03-26:
I think Serial module should work theoretically fine with Rpi as long as connections are fine. The real trick is modifying the libmodbus code to make it work properly.

rpiModbus, 2015-03-26:
@Chadster766 i have the 3204MSO, it has also many digital inputs very useful in cases like this with many digital signals to monitor.

@dhruvvyas90 the part of code that is working is this:

enum {
    TCP,
    RTU
};

int main(int argc, char *argv[])
{
    int socket;
    modbus_t *ctx;
    modbus_mapping_t *mb_mapping;
    int rc;
	int i;
	uint16_t tab_reg[64];
	

	printf("Waiting for RTU Serial connection\n");
	
        ctx = modbus_new_rtu("/dev/ttyAMA0", 9600, 'N', 8, 1);
		
		modbus_set_debug(ctx,TRUE);    //Enable debugging for tests
        //modbus_set_slave(ctx, 1);
		if (modbus_set_slave(ctx,1)== -1)
			{
			fprintf(stderr, "set slave failed: %s\n", modbus_strerror(errno));
			modbus_free(ctx);
			return -1;
			}
		
		//istruzioni rts
		modbus_rtu_set_serial_mode(ctx, MODBUS_RTU_RS485);
        modbus_rtu_set_rts(ctx, MODBUS_RTU_RTS_UP);
		printf("prima di connect \n");
        //modbus_connect(ctx);
		if (modbus_connect(ctx) == -1)
			{
				printf("modbus_connect(ctx) == -1\n");
			fprintf(stderr, "Connection failed: %s\n", modbus_strerror(errno));
			modbus_free(ctx);
			return -1;
			}
		printf("letture riga 67\n");
		rc = modbus_read_registers(ctx, 1, 0x0000, tab_reg);
		
		if (rc == -1) {
			printf("rc == -1 \n");
			fprintf(stderr, "%s\n", modbus_strerror(errno));
			return -1;
		}
	     
		printf("Serial connection started!\n");
		
		for (i=0; i < rc; i++) {
			printf("reg[%d]=%d (0x%X)\n", i, tab_reg[i], tab_reg[i]);
			}
			
    

    mb_mapping = modbus_mapping_new(MODBUS_MAX_READ_BITS, 0,                                  MODBUS_MAX_READ_REGISTERS, 0);
    if (mb_mapping == NULL) {
        fprintf(stderr, "Failed to allocate the mapping: %s\n",
                modbus_strerror(errno));
        modbus_free(ctx);
        return -1;
    }

the table of addresses is this:

[image]

thanks

rpiModbus, 2015-03-26:
i'm trying the receiver stage, i've connected RE and A to GND and B to +3.3, on RO i have a square wave signal with 2kHz frequency,is it normal?

rpiModbus, 2015-03-26:
my fault, the supply was not enough

stephane, 2015-05-21:
It would be very great if someone could take time to write a documentation about libmodbus on Raspberry Pi because there are many questions about this subject.

Anyone is interested to do that in this thread?

You just need to share a bit of your experience by creating a page dedicated to Raspberry Pi on the wiki:
https://github.com/stephane/libmodbus/wiki

closed 2015-05-21 by stephane