stephane/libmodbus #254: Running on windows

thaxy, 2015-03-27:
Hello,

I compiled version 3.1.1.2. I know it is unstable.
But is this working on 3.0.6? Couldn't find any open issue related to that.

[image]
[image]

```c
    modbus_t *ctx = modbus_new_rtu("COM5", 9600, 'E', 8, 1);

	modbus_set_debug(ctx, TRUE);
	if (modbus_set_slave(ctx, 1) == -1)
	{
		fprintf(stderr, "set slave failed: %s\n", modbus_strerror(errno));
		modbus_free(ctx);
		return -1;
	}
	modbus_rtu_set_serial_mode(ctx, MODBUS_RTU_RS485);
	modbus_rtu_set_rts(ctx, MODBUS_RTU_RTS_UP);

	if (modbus_connect(ctx) == -1)
	{
		printf("modbus_connect(ctx) == -1\n");
		fprintf(stderr, "Connection failed: %s\n", modbus_strerror(errno));
		modbus_free(ctx);
		return -1;
	}

	uint16_t tab_reg[10];
	int rc = modbus_read_registers(ctx, 1, 0x0000, tab_reg);

	if (rc == -1) {
		printf("rc == -1 \n");
		fprintf(stderr, "%s\n", modbus_strerror(errno));
		return -1;
	}
```

thaxy, 2015-03-27:
Damn. Same problem with 3.1.2.2. Anyone an idea what am I doing wrong?

closed 2015-03-27 by thaxy

meongit, 2015-12-04:
How did you fix this?

meongit, 2015-12-04:
There was no error. There was no server running hence the problem.