stephane/libmodbus #256: noise over A and B

rpiModbus, 2015-04-01:
Hi, here i am again, i had some progress, but i can receive a good answer to a request every two or three times. in the attachments there are the signals on A and B, someone could help me? how can i reduce this noise? thanks
[image]
[image]
[image]
[image]

dhruvvyas90, 2015-04-01:
Use a terminating resistor of 100 ohms at RS485 side. It might work.

rpiModbus, 2015-04-02:
hi, thanks for your answer, but nothing gets better. could be the supply of the transceiver at 3.3v and the bus communication at 5v? if you see in the first fig. the transmission is at 3.3 and the answer at 5v

stephane, 2015-05-21:
You problem is very interesting but the github issue tracker isn't the right place to talk about that (see libmodbus mailing list).

closed 2015-05-21 by stephane