stephane/libmodbus #257: RTU over TCP support in master branch

xabies, 2015-04-15:
Hi, I want to use this library but I need support for RTU over TCP too. I have found a patch for this in:
https://groups.google.com/forum/#!msg/libmodbus/xm8S6MMDhgM/Qd97y9_mhwgJ

This patch works fine but is for the version 3.0.3. I have tried to apply this patch to the master branch but I have not been able to compile the master branch with these cahnges. There are a few little changes but I suppose that something more has change in these new version.

Anyone could help me? Thanks.

xabies, 2015-04-15:
I was thinking... other possible solution could be use the branch 3.0.3 and apply the patch to this branch. The problem is that i will need to open the proyect with visual studio 208. The branch master has some files to open it with visual studio and build a .ddl and a .lib; but in 3.0.3 branch this option is not available.

kiug, 2015-04-16:
Hi, it would be nice if this functionality was included in the official version.
In my practice, I met with a number of cases where the device is connected through NPort (http://www.moxa.com/product/SDS_SerialDevice%20Servers.htm).

stephane, 2015-04-22:
@xabies new features must be added to master branch only So if you want to work on it, you must create a branch of master and send a PR (pull request) for review when it's ready.

xabies, 2015-04-27:
Hi Stephane. This will be great, but unfortunately I am a newbie and I think I am not able to do that. I found this patch and I think that this feature could be interesting for more people. This patch  breaks the design guidelines of the beautiful original code, so if someone could rewrite this patch for the current version of libmodbus in a proper way everybody colud enjoy this new feature.

There are only a few of changes in the patch, so I think that this will not be difficult for a C developer.

Thnaks and congratiolations for your job.

henkhoo, 2017-12-18:
i would like to use RTU over TCP. ( if possible with check_modbus nagios plugin )
i understand this is not possible with undelaying libmodbus ,  now i read here about patch ( txt file )
( https://groups.google.com/forum/#!msg/libmodbus/xm8S6MMDhgM/Qd97y9_mhwgJ )

but where to find how to install/use such a patch for a noob.

alorle, 2022-11-25:
Do we know anything else about this? I mean, we've been around this for seven years... For a request that has the code done (may require work -styling, cleanup, testing-) and has been claimed multiple times, that seems like a long time, right? With all my respect, @stephane