stephane/libmodbus #258: libmodbus RTU ignoring slave address in response

KGrizz, 2015-04-19:
I think I've found a bug in this library in RTU mode. When libmodbus receives a respond after a request, it isn't checking whether that request was sent by a desired slave.
It happens for example when slave sends the response too late while libmodbus issued another request with identical data size and register address but different slave address.
Is it a bug or am I missing something?
I'm using version 3.0.3-1 on Raspberry Pi 2 with USB to RS485 converter.

svisalakshi8, 2015-04-23:
Hi, I am running my libmodbus rtu slave code in my Raspberry pi and running my modbus rtu master code in windows. The request from the master is not received by the code correctly. Can u please help me on that.

kingstien, 2015-04-30:
I dont know whether the version 3.0.3-1 check the slave address,but the version 3.1.2 did.

KGrizz, 2015-04-30:
I've compared source codes of current and Raspbian version of libmodbus and
voila - all I need to do is compile current version by my self. The older
one wasn't comparing addresses.

Thank you for your help.

closed 2015-05-05 by stephane