stephane/libmodbus #259: Modbus RTU not working

svisalakshi8, 2015-04-22:
I am running libmodbus( libmodbus-3.1.2) modbus rtu slave code in my Raspberrypi and modbus master simulator running in my windows pc.  I have connected two pc via usb to rs232 converter cable.When i send request from the master simulator(running on windows) the slave code receives it wrongly.
i.e.
i am sending 01 03 00 00 00 0A C5 CD
but i am receiving it as 01 03 00 00 00 0A 05 1E

I also tried changing the values at the sender side and verify, what i found is that if i give value below 1F it is received properly at the slave side and any value above 1F is coming wrongly. Please help

kingstien, 2015-04-30:
Does both side have the same DataBit and StopBit?

stephane, 2015-05-21:
@svisalakshi8 could you please use the libmodbus mailing list for questions.

closed 2015-05-21 by stephane