stephane/libmodbus #262: modbus_read_registers - change *dest to signed integer

davisjp1822, 2015-05-07:
Hello - thank you for the fantastic library!

I am using libmodbus to control an AMCI SMD23E integrated stepper/motor drive over Modbus/TCP:

http://amci.com/stepper-motor-control/integrated-stepper-motor-control-smd23e.asp

The SMD reports motor position and encoder position via the Input Words. These values are reported in 16 bit words (decimal) that can be negative. Currently, *dest in modus_read_registers is a uint16_t - not allowing me to read negative values. Could you change this to int16_t please?

Thank you!

mhei, 2015-05-07:
Isn't a simple type cast a solution - or do I miss anything?

```c
int16_t position;
...

rc = modbus_read_registers(ctx, 0, 1, (uint16_t *)&position);
```

davisjp1822, 2015-05-08:
I am not sure that will work. CLANG gives me an error of, "incompatible pointer types passing "int32_t *" to parameter of type "uint16_t *".

mhei, 2015-05-08:
Please try it out, it will work. Modbus standard does not define the signess of the registers, as this is totally device specific. libmodbus provide functions to retrieve the registers, but leave it to the applications to intepret the data. So when your compiler complains about the casting, but you know that are doing the right thing, then disable the warning or use a workaround, e.g. by defining a union with two members (signed, unsigned) and then pass the correct pointer down to libmodbus while using the other one when accessing it in your application.

mhei, 2015-05-20:
Any news on this? If my suggested solution works, then please close this ticket. Thanks.

closed 2015-05-21 by stephane

davisjp1822, 2015-05-29:
My apologies for the delay - I got tied-up on another project. The cast works - thanks again for the help.