stephane/libmodbus #265: Access violation in function modbus_mask_write_register

lipengin1975, 2015-05-20:
Hi, everyone!
I use the modbus_mask_write_register function to write a bit value into a holding register.But it throw a exception in memory access.
I browse the code of version 3.1.2, in line 1471 of file modbus.c, there defined the req array like that:
uint8_t req[_MIN_REQ_LENGTH];
the macro _MIN_REQ_LENGTH value is 12, but in line 1480 of the modbus.c,
    req[req_length++] = and_mask >> 8;
    req[req_length++] = and_mask & 0x00ff;
    req[req_length++] = or_mask >> 8;
    req[req_length++] = or_mask & 0x00ff;
so I think at last req_length equals 14, is it a bug?

closed 2015-05-21 by stephane

stephane, 2015-05-21:
I merged it but a bit hastily, the fix allocates too much memory and this bug exists because there is no tests for this feature! So I'll improve that with another patch.

lipengin1975, 2015-05-26:
Thanks mhei and stephane !  I changed the code same as mhei's method, but now I think the better way is what stephane have done.

commit b7894da5 (2015-05-21): modbus_mask_write_register: fix buffer overflow (fixes #265)

Thanks to @lipengin1975 for reporting this issue.
commit 9b4212c8 (2015-05-21): modbus_mask_write_register: better fix with unit test (closes #265)