stephane/libmodbus #269: Possible segfault in modbus_reply and modbus_reply_exception

sstiller, 2015-06-10:
In both functions (modbus.c) there is a check if the pointer ctx is NULL. But some lines above, the pointer is used:
    int offset = ctx->backend->header_length;
This makes the test useless because the software already should have crashed.

stephane, 2015-06-11:
Could you provide a github link to the concerned lines, please?

sstiller, 2015-06-11:
modbus_reply:
Pointer ctx  used: https://github.com/stephane/libmodbus/blob/master/src/modbus.c#L683
pointer checked: https://github.com/stephane/libmodbus/blob/master/src/modbus.c#L691

modbus_reply_exception:
Pointer ctx used: https://github.com/stephane/libmodbus/blob/master/src/modbus.c#L1065
pointer checked: https://github.com/stephane/libmodbus/blob/master/src/modbus.c#L1073

stephane, 2015-06-11:
Thank you for your bug report.

commit 0e2f470a (2015-06-11): Fix late check of ctx in modbus_reply[|_exception] (closes #269)
closed 2015-06-11 by stephane