stephane/libmodbus #271: customize checksum method

lifsong, 2015-06-17:
Hi,  libmodbus authors,

We're using modbus on our arm linux app talk to our Soc app,  but the Soc app side cannot reserve enough memory for CRC16  tables.  Would you consider add a modbus api to allow users to inject customized checksum method? 

Another related minor question is that libmodus is logging to stdout/stderr, would you also consider allowing users to define where to write logs?

stephane, 2015-06-17:
Yes for both!

karlp, 2015-06-18:
I'm _reallly_ curious to hear about the environment you're working in where you're running linux, with full scale libraries, but can't afford the ram for the checksum, but dynamically loading another one would be ok?  I know people can come up with justifications for just about anything, I'm just very impressed at this one, the tables are only 512 bytes :)

The logging replacement would be totally great though

stephane, 2015-08-04:
Not an issue (just questions) so I close it.

closed 2015-08-04 by stephane