stephane/libmodbus #273: Seg Fault

xinthose, 2015-07-03:
Hello.  I keep on getting a segmentation fault when I try to set a response timeout.  Can anyone see why?  The code compiles.  Thank you.

int create_modbus_tcp_connection (void)
{
	int eth_conn_status_int;
	modbus_t *ctx;
        
	
	std::ifstream eth_conn_status ("/sys/class/net/eth2/carrier");	// closest to power switch	
	eth_conn_status >> eth_conn_status_int;	// 0 >> not connected || 1 >> connected
	if (eth_conn_status_int == 1) {ctx = modbus_new_tcp ("192.168.2.101", 502);}	// WebRelay address and default Modbus Port (502)
	else
	{
		std::cerr << "create_modbus_tcp_connection >> modbus_new_tcp >> ERROR >> ethernet cable is unplugged" << std::endl;
		return -1;
	}
	if (ctx == NULL)
	{
		std::cerr << "create_modbus_tcp_connection >> modbus_new_tcp >> ERROR >> " << modbus_strerror(errno) << std::endl;
		return -1;		
	}
	else
	{
		modbus_flush (ctx);
		modbus_set_response_timeout (ctx, 3, 0);	// *** SEG FAULT ***
	}	
	if (modbus_connect(ctx) == -1) 
	{
		std::cerr << "create_modbus_tcp_connection >> modbus_connect >> ERROR >> " << modbus_strerror(errno) << std::endl;
		modbus_free(ctx);
		return -1;
	}

	return 0;
}

karlp, 2015-07-03:
compiling with libmodbus 3.0 and running with libmobus 3.1? the signature changed for set_timeout at least?

xinthose, 2015-07-03:
I am using 3.0.6.  Is 3.1.2 safe to use?

xinthose, 2015-07-03:
I installed 3.1.2 and used that and now it doesn't seg fault anymore!  Thank you.

closed 2015-07-03 by xinthose