stephane/libmodbus #276: USB -> ch341 -> RS485?

snakesandstuff, 2015-07-22:
1. Version: 3.1.2 downloaded and compiled and installed from https://github.com/stephane/libmodbus/archive/master.zip
2. Run on a raspberry pi (4 core ARMv7)
3. RTU backend
4. Debug output:
pi@modbus ~ $ ./modbus2
Connected to /dev/ttyUSB0
Slave set to 0x01
[01][03][75][35][00][01][8E][08]
ERROR Bad file descriptor
Bad file descriptor

The code that is not working and giving the output listed above:
Code being used at: http://pastebin.com/2CLx6agc

Unit tests in test directory for TCP functions work properly

The following was used to compile the code:
gcc -I/usr/local/include/modbus -L/usr/local/lib -lmodbus modbus2.c -o modbus2

The USB->RS485 adapter works properly when tested by sniffing serial port as well as using modpoll on a Win7 machine.

Additional Information:

dmesg output when adapter is plugged into USB port:
[ 1296.027230] usb 1-1.3: new full-speed USB device number 4 using dwc_otg
[ 1296.130334] usb 1-1.3: New USB device found, idVendor=1a86, idProduct=7523
[ 1296.130359] usb 1-1.3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 1296.130377] usb 1-1.3: Product: USB2.0-Serial
[ 1296.156478] usbcore: registered new interface driver usbserial
[ 1296.156621] usbcore: registered new interface driver usbserial_generic
[ 1296.156740] usbserial: USB Serial support registered for generic
[ 1296.160359] usbcore: registered new interface driver ch341
[ 1296.160514] usbserial: USB Serial support registered for ch341-uart
[ 1296.160620] ch341 1-1.3:1.0: ch341-uart converter detected
[ 1296.164736] usb 1-1.3: ch341-uart converter now attached to ttyUSB0

gcc -v output:
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.6/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Debian 4.6.3-14+rpi1' --with-bugurl=file:///usr/share/doc/gcc-4.6/README.Bugs --enable-languages=c,c++,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.6 --enable-shared --enable-linker-build-id --with-system-zlib --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.6 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --enable-plugin --enable-objc-gc --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.6.3 (Debian 4.6.3-14+rpi1)

Any help/insight is greatly appreciated.

Thank you
Mail Delivery Subsystem <mailer-daemon@google.com>
	
3:52 PM (6 hours ago)
		
to me
Bonjour snakesandstuff@gmail.com,

Il se peut que le groupe que vous avez essayé de contacter (libmodbus) n'existe pas ou que vous ne soyez pas autorisé à lui envoyer des messages. Voici quelques-unes des raisons possibles pour lesquelles votre message n'a pas pu être envoyé :

 * Vous avez mal orthographié le nom du groupe.
 * Le propriétaire du groupe l'a peut-être supprimé.
 * Vous devez être abonné avant de pouvoir envoyer des messages.
 * Ce groupe n'autorise pas l'envoi de message.

Si vous avez des questions sur ce groupe ou tout autre groupe Google, consultez le centre d'aide à l'adresse http://groups.google.com/support/.

Merci,

Google Groups

----- Original message -----

X-Received: by 10.152.1.105 with SMTP id 9mr799577lal.3.1437511946177;
        Tue, 21 Jul 2015 13:52:26 -0700 (PDT)
Return-Path: <snakesandstuff@gmail.com>
Received: from mail-wi0-x230.google.com (mail-wi0-x230.google.com. [2a00:1450:400c:c05::230])
        by gmr-mx.google.com with ESMTPS id bk2si602546wib.1.2015.07.21.13.52.26
        for <libmodbus@googlegroups.com>
        (version=TLSv1.2 cipher=ECDHE-RSA-AES128-GCM-SHA256 bits=128/128);
        Tue, 21 Jul 2015 13:52:26 -0700 (PDT)
Received-SPF: pass (google.com: domain of snakesandstuff@gmail.com designates 2a00:1450:400c:c05::230 as permitted sender) client-ip=2a00:1450:400c:c05::230;
Authentication-Results: gmr-mx.google.com;
       spf=pass (google.com: domain of snakesandstuff@gmail.com designates 2a00:1450:400c:c05::230 as permitted sender) smtp.mail=snakesandstuff@gmail.com;
       dmarc=pass (p=NONE dis=NONE) header.from=gmail.com
Authentication-Results: mx.google.com;
       dkim=pass header.i=@gmail.com
Received: by mail-wi0-x230.google.com with SMTP id gb10so70032630wic.1
        for <libmodbus@googlegroups.com>; Tue, 21 Jul 2015 13:52:26 -0700 (PDT)
DKIM-Signature: v=1; a=rsa-sha256; c=relaxed/relaxed;
        d=gmail.com; s=20120113;
        h=mime-version:date:message-id:subject:from:to:content-type;
        bh=YhNGlWhPL68cbxO49EucFQq/OxHUMyx7uq1F8qalM9Y=;
        b=JxHivtMOqI3eYxhcxTWvBsSGB805pDQoETLzkOD7ehqzFqsEhiLUNRT7ztw2tm+STW
         NuHJrHfjniWNvoYm5A/Us4ygIG8fheIct0KHdiB77VIb8suzOzJQ+ypcvB1wKz1jQu5d
         LgaxFDWC3jIlwOhBDkT1IrcZE21CPR4GrWMkFGG1AVGI2KaeWCNM+xFoa6Ue4v7Vvxjl
         oMNCrpVzIk8+8/vqDj0Cm7l9qhoZKDw10ZrHM5fABKQG6jN5LDY7T9tXTp/qgJs+gCnA
         C7IUopk/uIsSSWegH0Rd868K6Z+unIxoMoqucYiMLf8zvO/MY46tjk/D/9MIkf1RT8oa
         3xIg==
MIME-Version: 1.0
X-Received: by 10.180.75.243 with SMTP id f19mr23034735wiw.52.1437511946033;
 Tue, 21 Jul 2015 13:52:26 -0700 (PDT)
Received: by 10.28.146.148 with HTTP; Tue, 21 Jul 2015 13:52:25 -0700 (PDT)
Date: Tue, 21 Jul 2015 15:52:25 -0500
Message-ID: <CAB4RPXwVw4MEtFB+tMO3gK9uj0caYAW=KO5T4uq-zQzujC2qaA@mail.gmail.com>
Subject: Problems using USB->RS485 Adapter?
From: Bobby Neal <snakesandstuff@gmail.com>
To: libmodbus@googlegroups.com
Content-Type: multipart/alternative; boundary=f46d0438951be75f3c051b68d48d

1. Version: 3.1.2 downloaded and compiled and installed from
https://github.com/stephane/libmodbus/archive/master.zip
2. Run on a raspberry pi (4 core ARMv7)
3. RTU backend
4. Debug output:
pi@modbus ~ $ ./modbus2
Connected to /dev/ttyUSB0
Slave set to 0x01
[01][03][00][00][00][01][84][0A]
ERROR Bad file descriptor
Bad file descriptor

Have other had problem when using a ch341 based USB->RS485 adapter? Perhaps there is a driver problem?

The code that is not working and giving the output listed above:
Code being used at: http://pastebin.com/HvUyHcUv

Thank you for any input.

karlp, 2015-07-22:
If you really want function code 3, You probably don't actually want register 30005, you probably want read_input_register(4).  but that's your world, and maybe it's correct.  Either way, that's not what's causing your bad file descriptor problems :)

the ch341 linux driver is wayyyyy buggy.  You will continue to have odd pain with it.  It "normally" works with no parity and a fixed baud rate, but any opening/closing of ports can leave it in undefined states.  You _can_ try hacking around with the vendor driver, and I started some work to clean it up, https://github.com/karlp/linux/commits/ch341-3.18.6 but it's a long road.

snakesandstuff, 2015-07-22:
Thank you for the response. I did notice that error after I posted, and you
are correct about the addressing.

Have you tested and used libmodbus with a USB->RS485 adapter of any type?
Are the FTDI type better?

Thank you again,

Bobby Neal

karlp, 2015-07-22:
Well, these ones work pretty well http://www.ftdichip.com/Products/Cables/USBRS485.htm and come cabled up.  I'm sure you can get others in different form factors.  cp210x devices would work fine too, and probably pl2303, but ftdi and cp210x are the ones I know will work.

karlp, 2015-07-22:
(I presume you have one of these: http://www.ebay.com/itm/New-USB-to-RS485-USB-485-Converter-Adapter-Support-Win7-XP-Vista-Linux-Mac-OS-/381097954687?pt=LH_DefaultDomain_0&hash=item58bb354d7f  they're really common, and really janky)

snakesandstuff, 2015-07-22:
You are correct. I have one of the ftdi's on order now. I think in the mean time (while waiting on delivery) I will play around and see exactly where in the code the error is coming from and see if it is perhaps with trying to set RTS or something else using ioctl that the ch341 doesn't understand.

I have tested the ch341 under this environment with simple send and receive using echo "Test" > /dev/ttyUSB0 and listening for test messages from another device with cat /dev/ttyUSB0

karlp, 2015-07-22:
Yeah, unless you _realllly_ know that are using the RTS pin explicitly for rs485 driver enable, just don't use that part of libmodbus.  Likewise for things like "set rs485 mode"  that stuff only does what you want  if you have a serial port that the kernel itself understands rs485 mode for.  With transparent devices like the ftdi cables and those ch34x dongles, there's no rts signal being used.  FTDI has a native DE pin, and the ch340 dongle I have uses nand gates to do an "auto" mode switch, based on an old circuit cellar article.  In no case do you even tell libmodbus about any of that.

snakesandstuff, 2015-07-22:
I feel completely silly... In my code I never used a modbus_connect, so the file descriptor was never being set properly... Just posting in case someone else has a similar problem.

karlp, 2015-07-22:
I didn't notice that either, and honestly, I feel that libmodbus should provide a better error message in that case.  @stephane opinions?

stephane, 2015-08-04:
When I was reading the comments of the issue, I found funny you don't talk about the missing connect() call. Hopefully, you found it not too long after you open the issue ;)

I just looked at source code to help the user about this mistake but I don't find a good place to handle that so I edited the documentation to avoid this mistake.

closed 2015-08-04 by stephane