stephane/libmodbus #278: modbus_report_slave_id() timeout

epsilonrt, 2015-07-27:
Hello,
I use version 3.0.1 of Ubuntu 12.04 libmodbus in RTU mode to test a
Modbus slave under development.

libmodbus works great for read and write registers.
But I have a problem with the modbus_report_slave_id () that indicates a
timeout.
Yet the response timeout is set to 1 second and I can see that the
oscilloscope response from the slave is correct and sent very quickly.
In fact the answer is displayed in debug mode by libmodbus.
That is what my program displays:
----------------------------
mbpoll(debug): debug enabled
mbpoll(debug): Set mode to RTU for serial port
mbpoll(debug): Set device=/dev/ttyUSB1
mbpoll 1.0-52 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright © 2015 Pascal JEAN aka epsilonRT
All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyUSB1 at 19200 bauds (E, 8, 1)
mbpoll(debug): Byte timeout: 0 s, 500000 us
mbpoll(debug): Set timeout to 1 sec, 0 us
Protocol configuration: Modbus RTU
Slave configuration...: address = 32, report slave id
Communication.........: /dev/ttyUSB1, 19200-8E1
                        t/o 1.00 s, poll rate 1000 ms
[20][11][D8][7C]
Waiting for a confirmation...
ERROR Connection timed out: select
Report slave ID failed(-1): Connection timed out
<20><11><10><FF><68><75><6D><2D><31><2E><30><2D><35><35><02><04>

----------------------------
I tried with version 3.1.2, installed from the source code and the
problem is identical.
I do not see the problem.
Would you have any idea?
Thank You
Regards
Pascal

Ketorin, 2015-08-24:
Well, having superficially looked into that code lately to address my own issue, I'd say the simplest reason for this could be your slave isn't sending as many bytes of data as it's claiming in the first byte of the message body.

My little eye is spying that, assuming that's the whole ADU, your message body excluding the size byte is only 10 bytes, while lenght field tells libmodbus to expect 16.

stephane, 2015-08-27:
Thank you for your answer @Ketorin

closed 2015-08-27 by stephane