stephane/libmodbus #280: modbus_receive_confirmation() Does not Receive the Full Response

Ketorin, 2015-08-21:
This may be known issue... May even be how it's supposed to work.

Anyway, I'm trying to implement modbus "find file record" and "read file record" using modbus_send_raw_request() and modbus_receive_confirmation(). As you may remember, these modbus commands are a bit special and the internal receive_msg() function seems not to be able to figure out the message lenght correctly, it returns a bogus value, 9 which with modbus TCP would include the FN code and the byte after that.

I also tried to call modbus_receive_confirmation() another time, with my receive butffer + original read lenght. This time modbus_receive_confirmation() returned a value that was the expected amount of bytes but no new values were written to the receive buffer.

mhei, 2015-08-21:
I guess that compute_data_length_after_meta and/or compute_meta_length_after_function return invalid lengths for your request/function code and thus you have to tweak this...

Ketorin, 2015-08-21:
Ah, I turned up the debug, actually the second call to modbus_receive_confirmation() reads the rest of the buffer accordingly, but terminates into read error before values get copied to my receive buffer.

Is it kosher to call modbus_receive_confirmation() more than once?

Ketorin, 2015-08-21:
OK so,  length_to_read = ctx->backend->header_length + 1; on line 355 on modbus.c seems to be what keeps my code chugging along*, and also answers the question should modbus_receive_confirmation() be called multiple times. Apparently not.

Well, it seems I may be able hack around this by setting the field by hand before calling modbus_receive_confirmation() the second time...

* I tried to call "write file record" and as it happens, the response packet is just a byte shorter of what libmodbus is trying to read, if I get it right.

Ketorin, 2015-08-24:
Well, thank you, your help was spot-on.

I ended up implementing the two commands in a simplified way; multi-record commands would call for two-dimensional source and destination arrays and be a complete aversion from the rest of the api. Maybe, if the functions were made variadic, but that would be a bit ugly too. You'd need either to tell the number of buffers or have the last one be NULL pointer.

I tried to make a diff, just in case you are interested. (Plus there's this whole LGPL thing...)
http://www.pastebin.ca/3125922

Because I'm mapping these into some string registers at the receiving end, there's the special case of lenght 0xffff, which in my implementation tells the slave to return the whole register content. This also means you can't calculate the response lenght from the request; so even if this does not break the letter of the standard, it certainly does violate its spirit.

stephane, 2015-08-27:
Too bad the patch is against libmodbus 3.0.6 and not master branch. Do you want to merge it upstream?

Ketorin, 2015-08-28:
Hmm. You could make a three-way diff? To get a diff against the master.

What do you think about the multiple records support with commands 20 and 21? Should libmodbus concern it and if so, should it be two dimensional array or variadic function?

I could try to, eventually.

stephane, 2015-11-16:
I tried to create a new branch to manually apply your patch against but it's too painful w/o full source files.
Could you fork/PR as usual, please?

Ketorin, 2015-11-20:
I tried to send you a pull request with my changes against version 3.1.2.

Please let me know if I'm doing it wrong.

abammatter, 2015-12-09:
I'm experiencing the same problems using v3.1.2
In my case modbus_receive_confirmation() returns 6 Bytes while I'm expecting 8 Bytes to read back. With a logic analyzer attached to my board I figured out that RTS is put back to write mode after the sixth Byte arrived. In modbus-rtu-private.h is a define _MODBUS_RTU_PRESET_REQ_LENGTH  6 which I am suspecting to cause this error, since I could not find any time variable while in reading mode.
Can someone confirm this issue and am I on the right track?

stephane, 2016-05-18:
@Ketorin is it this PR https://github.com/stephane/libmodbus/pull/299 ?

MinimumLaw, 2016-05-26:
Hi!

I think PR #330 fix this (for Read File Record/Write File Record - 0x14 and 0x15), but what about Find File Record? Is this FC part of specification or additional command?