stephane/libmodbus #282: Custom Allocator Support

Ketorin, 2015-08-26:
Would you think this is worth while?

I actually have an implementation that is enough for my personal needs, meaning the pointers to the custom allocators are static. This is because modbus_new_rtu, modbus_new_tcp and modbus_new_tcp_pi return a pointer that they allocate themswelves, it's a bit untrivial to implement correcly, unless I made a second version of every function that takes a custom malloc, but that doesn't feel right.

If you are interested, I could modify my implementation to please your needs and try to correctly do this whole fork and pull-request thing.

stephane, 2015-08-27:
Could you explain me the goal of the this custom allocator?

Ketorin, 2015-08-28:
I'm using a custom heap, which is capped to some fairly small amount. Also it logs, if the loglevel is at the highest. When it has leaked all its memory, it will throw everything away and start from the cratch, hopefully in a graceful manner. Won't even cause too much more roll-back cemantics, because I'm doing that already so not to lose state at power-loss; you, it wont be running in some fancy datacenter with proper resource capping and UPS, mere some junky windows 2000 toaster that just has been lying around.

I surely don't know if this is a standard thing.

karlp, 2015-08-28:
if you reallllllllly need this, the way libevent2 does it might be easier and friendlier to implement and to use..  http://www.wangafu.net/~nickm/libevent-book/Ref1_libsetup.html  search down for the section titled "memory management"

stephane, 2015-10-06:
Not sure it's a good idea and such a feature will complicate the code.

closed 2015-10-06 by stephane