stephane/libmodbus #285: winsock2.h : present but cannot compiled

dragoshenron, 2015-09-14:
Dear all,
I have just installed the library (3.1.2) under mingw32/msys.

When launching ./configure, I got the following error message:
checking winsock2.h usability... no
checking winsock2.h presence... yes
configure: WARNING: winsock2.h: present but cannot be compiled
configure: WARNING: winsock2.h:     check for missing prerequisite headers?
configure: WARNING: winsock2.h: see the Autoconf documentation
configure: WARNING: winsock2.h:     section "Present But Cannot Be Compiled"
configure: WARNING: winsock2.h: proceeding with the compiler's result
configure: WARNING:     ## ----------------------------------------------------------- ##
configure: WARNING:     ## Report this to https://github.com/stephane/libmodbus/issues ##
configure: WARNING:     ## ----------------------------------------------------------- ##

I am running Windows Vista Professional (and I can't upgrade to another OS).

Any suggestions?

Thanks
Alex

stephane, 2015-10-06:
Did you read https://github.com/stephane/libmodbus/tree/master/src/win32?

dragoshenron, 2015-10-09:
Dear Stephane,
Yes, indeed I read that link. However, I'm not using VS.
Please note that I am able to develop, compile and run my software simply with MinGW/MSYS and plain gcc.
This is not a bug report but I received a warning from the ./configure to report this issue... so I did.

Regards,
Alex

stephane, 2015-10-09:
so you have been able to compile properly?
If so, you can just ignore this warning, BTW I don't know how to fix it

dragoshenron, 2015-10-10:
Yes, indeed I was able to compile it properly and running my own code.

I'm sorry I can't help you either to fix this issue as I'm not skilled in the autoconf/automake world.

It's going without saying that you can close this issue.

stephane, 2016-05-18:
OK thank you.

closed 2016-05-18 by stephane