stephane/libmodbus #287: RTU unit tests return connection reset by peer Ubuntu

ukmeterman, 2015-10-05:
Hi,
I am trying to use the modbus library in RTU mode, under Ubuntu 14.04.3  I am finding that libmodbus 3.0.1 3.0.6 is giving the error connection reset by peer, even for the client server unit tests. What am I doing wrong? 

Hardware is true pci serial cards, and the connection is fine useing terminal to check the null modem cable.  Connected to com4 and com5, hardware works under windows 7

stephane, 2015-10-06:
Could you explain how you launch the tests (command line, host, etc)?

ukmeterman, 2015-10-07:
Hi,
Thanks for getting back to me 
./unit-test-server rtu 
ERROR Connection reset by peer: read
Opening /dev/ttyS5 at 115200 bauds (N, 8, 1)
Waiting for a indication...
Quit the loop: Connection reset by peer

./unit-test-client rtu
** UNIT TESTING **

TEST WRITE/READ:
[11][05][00][13][FF][00][7F][6F]
Waiting for a confirmation...
ERROR Connection reset by peer: read
Opening /dev/ttyS4 at 115200 bauds (N, 8, 1)
1/2 modbus_write_bit: FAILED

Version output:
Compiled with libmodbus version 3.0.6
Linked with libmodbus version 3.0.6
The functions to read/write float values are available (2.1.0).
Oh gosh, brand new API (2.1.1)!

Configure output
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by libmodbus configure 3.0.6, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ ./configure 

## --------- ##
## Platform. ##
## --------- ##

hostname = james-P5K
uname -m = x86_64
uname -r = 3.13.0-65-generic
uname -s = Linux
uname -v = #105-Ubuntu SMP Mon Sep 21 18:50:58 UTC 2015

/usr/bin/uname -p = unknown
/bin/uname -X     = unknown

/bin/arch              = unknown
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /home/james/CodeSourcery/Sourcery_CodeBench_Lite_for_Nios_II_GNU_Linux/bin/
PATH: /home/james/CodeSourcery/Sourcery_CodeBench_Lite_for_Nios_II_GNU_Linux/bin
PATH: /usr/local/sbin
PATH: /usr/local/bin
PATH: /usr/sbin
PATH: /usr/bin
PATH: /sbin
PATH: /bin
PATH: /usr/games
PATH: /usr/local/games

## ----------- ##
## Core tests. ##
## ----------- ##

configure:2506: checking for a BSD-compatible install
[... 1822 lines cut ...]
checking for socket... yes
checking for strerror... yes
checking for strlcpy... no
checking for getaddrinfo... yes
checking winsock2.h usability... no
checking winsock2.h presence... no
checking for winsock2.h... no
checking whether TIOCSRS485 is declared... yes
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating src/Makefile
config.status: creating src/modbus-version.h
config.status: creating doc/Makefile
config.status: creating tests/Makefile
config.status: creating libmodbus.pc
config.status: creating libmodbus.spec
config.status: creating config.h
config.status: config.h is unchanged
config.status: creating tests/unit-test.h
config.status: tests/unit-test.h is unchanged
config.status: executing depfiles commands
config.status: executing libtool commands

Make output

stephane, 2016-01-15:
I'm pretty sure it's just a permission issue and no answer from reporter.

closed 2016-01-15 by stephane