stephane/libmodbus #292: MinGw: Warning on ./configure , error on make

SebastianBT, 2015-10-22:
Hello,

I try to compile libmodbus under MinGw and I have a bunch of warnings and it does not compile when running make.

--------------------------------------------------------------------------------------------------------------

When running ./configure:

configure: WARNING: The tools to build the documentation aren't installed
configure: WARNING: winsock2.h: present but cannot be compiled
configure: WARNING: winsock2.h:     check for missing prerequisite headers?
configure: WARNING: winsock2.h: see the Autoconf documentation
configure: WARNING: winsock2.h:     section "Present But Cannot Be Compiled"
configure: WARNING: winsock2.h: proceeding with the compiler's result
configure: WARNING:     ## ----------------------------------------------------------- ##
configure: WARNING:     ## Report this to https://github.com/stephane/libmodbus/issues ##
configure: WARNING:     ## ----------------------------------------------------------- ##
conftest.c:30: warning: left shift count >= width of type
conftest.c:30: warning: left shift count >= width of type
conftest.c:31: warning: left shift count >= width of type
conftest.c:31: warning: left shift count >= width of type
conftest.c:30: warning: left shift count >= width of type
conftest.c:30: warning: left shift count >= width of type
conftest.c:31: warning: left shift count >= width of type
conftest.c:31: warning: left shift count >= width of type
cc1.exe: warning: command line option "-fno-rtti" is valid for C++/ObjC++ but not for C
configure:12737: WARNING: The tools to build the documentation aren't installed
conftest.c:72: warning: conflicting types for built-in function 'memset'
conftest.c:92: warning: integer overflow in expression
/usr/lib/gcc/i686-pc-msys/3.4.4/../../../../include/w32api/winsock2.h:103:2: warning: #warning "fd_set and associated macros have been defined in sys/types.      This may cause runtime problems with W32 sockets"
/usr/lib/gcc/i686-pc-msys/3.4.4/../../../../include/w32api/winsock2.h:103:2: warning: #warning "fd_set and associated macros have been defined in sys/types.      This may cause runtime problems with W32 sockets"
configure:16880: WARNING: winsock2.h: present but cannot be compiled
configure:16880: WARNING: winsock2.h:     check for missing prerequisite headers?
configure:16880: WARNING: winsock2.h: see the Autoconf documentation
configure:16880: WARNING: winsock2.h:     section "Present But Cannot Be Compiled"
configure:16880: WARNING: winsock2.h: proceeding with the compiler's result

--------------------------------------------------------------------------------------------------------------
When running make:

make --no-print-directory all-recursive
Making all in src
  CC     modbus.lo
In file included from modbus.c:22:
modbus.h:16:20: stdint.h: No such file or directory
modbus.c: In function `_sleep_response_timeout':
modbus.c:104: warning: implicit declaration of function `nanosleep'
modbus.c:104: warning: nested extern declaration of `nanosleep'
make[2]: *** [modbus.lo] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

---------------------------------------------------------------------------------------------------------------------
Any Idea what is going wrong?

Thank you very much,

Best Regards,

Sebastian

P.S

Because I have no write-permissions I can not add the full logs.

stephane, 2015-10-22:
Which version of libmodbus?

Does it mean the _WIN32 flag is unset?
https://github.com/stephane/libmodbus/blob/master/src/modbus.c#L95

SebastianBT, 2015-10-22:
Hello Stephane,

in which file, or how can I figure out whether it is set or not?

Thank you very much,

Best Regards,

Sebastian

P.S.

 I just started with the libmodbus and MinGw, so sorry if I may ask some beginner's questions.

SebastianBT, 2015-10-22:
Hello Stephane,

I just put a check in the modbus.c a check:

#ifndef _WIN32
    #warning it is is not definden
#endif

_WIN32 is not defined.

Do you have any idea how I can solve the Problem in MinGw, or is the Problem in the libmodbus?

Thank you very much,

Best Regards,

Sebastian

SebastianBT, 2015-10-22:
P.S The Version is the "Head"(most curent Version) on Gitbub, but it also fails with the 3.1.2

SebastianBT, 2015-10-22:
Ok, I guess there are more problems:

I executed with C:\MinGW\msys\1.0\mysys.bat (shell) the commands:

1) autogen.sh
2) ./configure
3) make

I figured out that it may use another compiler than "MinGw", so I executed:

C:\MinGwLibmodbus\libmodbus-master>mingw32-make.exe

And now I got a different result:

mingw32-make.exe --no-print-directory all-recursive
Making all in src
  CC     modbus.lo
In file included from <command-line>:0:0:
./../config.h:254:17: error: duplicate 'unsigned'
 #define uint8_t unsigned char
                 ^
./../config.h:254:26: error: two or more data types in declaration specifiers
 #define uint8_t unsigned char
                          ^
./../config.h:246:18: error: duplicate 'unsigned'
 #define uint16_t unsigned short int
                  ^
./../config.h:246:27: error: duplicate 'short'
 #define uint16_t unsigned short int
                           ^
./../config.h:250:18: error: duplicate 'unsigned'
 #define uint32_t unsigned int
                  ^
Makefile:380: recipe for target 'modbus.lo' failed
mingw32-make[2]: *** [modbus.lo] Error 1
Makefile:393: recipe for target 'all-recursive' failed
mingw32-make.exe[1]: *** [all-recursive] Error 1
Makefile:278: recipe for target 'all' failed
mingw32-make.exe: *** [all] Error 2

Got anyone an idea what I should do? Which Tools need to be used to compile it? I read other people could do it, so what am I doing wrong? Is there another shell wor tool in MinGw to run autogen.sh and ./configure ?

Thank you very much,

Best Regards,

Sebastian

stephane, 2015-10-27:
I think I must play again with mingw to test it.

stephane, 2015-11-16:
Related to #285

SebastianBT, 2015-11-16:
I am out of the office until 01.12.2015.

Note: This is an automated response to your message  "Re: [libmodbus] MinGw: Warning on ./configure , error on make (#292)" sent
 on 11/16/2015 10:05:39 PM.

This is the only notification you will receive while this person is away.

Please consider the environment before you print / Merci de penser à l'environnement avant d'imprimer / Bitte denken Sie an die Umwelt bevor Sie drucken Bombardier Transportation GmbH Vorsitzender des Aufsichtsrats / Chairman of Supervisory Board: Dr. Lutz
 Bertling Geschäftsführung / Management Board: Dieter John (Vorsitzender/Chairman), Dr. Susanne Kortendick, Matthias Keller, Dr. Daniel Perlzweig Sitz der Gesellschaft / Principal Office: Berlin Registergericht / Registration Court: Amtsgericht Charlottenburg,
 HRB 64838

SebastianBT, 2015-11-16:
I am out of the office until 01.12.2015.

Note: This is an automated response to your message  "Re: [libmodbus] MinGw: Warning on ./configure , error on make (#292)" sent
 on 11/16/2015 9:46:52 PM.

This is the only notification you will receive while this person is away.

Please consider the environment before you print / Merci de penser à l'environnement avant d'imprimer / Bitte denken Sie an die Umwelt bevor Sie drucken Bombardier Transportation GmbH Vorsitzender des Aufsichtsrats / Chairman of Supervisory Board: Dr. Lutz
 Bertling Geschäftsführung / Management Board: Dieter John (Vorsitzender/Chairman), Dr. Susanne Kortendick, Matthias Keller, Dr. Daniel Perlzweig Sitz der Gesellschaft / Principal Office: Berlin Registergericht / Registration Court: Amtsgericht Charlottenburg,
 HRB 64838

SebastianBT, 2015-11-19:
I am out of the office until 01.12.2015.

Note: This is an automated response to your message  "Re: [libmodbus] MinGw: Warning on ./configure , error on make (#292)" sent
 on 11/18/2015 10:58:31 PM.

This is the only notification you will receive while this person is away.

Please consider the environment before you print / Merci de penser à l'environnement avant d'imprimer / Bitte denken Sie an die Umwelt bevor Sie drucken Bombardier Transportation GmbH Vorsitzender des Aufsichtsrats / Chairman of Supervisory Board: Dr. Lutz
 Bertling Geschäftsführung / Management Board: Dieter John (Vorsitzender/Chairman), Dr. Susanne Kortendick, Matthias Keller, Dr. Daniel Perlzweig Sitz der Gesellschaft / Principal Office: Berlin Registergericht / Registration Court: Amtsgericht Charlottenburg,
 HRB 64838

stephane, 2016-05-11:
The mingw env must be tested again for the final release of v3.2.0.
Help request on the mailing list.

closed 2016-05-11 by stephane