stephane/libmodbus #293: Modbus Connection Issue

mhdazeem, 2015-10-27:
Hi,

Really appreciate for the library. Actually I am implementing Modbus RTU in an embedded device Digi Module Connect me 9210. I am having the connection issue on both RTU and TCP Side. could you please help me out in this. I am using Digi ESP Net OS SDK for the implementation.

dhruvvyas90, 2015-10-27:
@mhdazeem What exactly is the problem ?

mhdazeem, 2015-10-27:
Thanks for the reply, When i try to make the connection with the slave using the " ctx = modbus_new_rtu("/dev/ttyUSB0", 19200, 'N', 8, 1);
    modbus_set_slave(ctx, SERVER_ID);" sample code its dosent make the connection keep saying connection failed.

mhdazeem, 2015-10-27:
server id is 1

karlp, 2015-10-27:
so... check errno, check you have perms on the serial port, check your return codes...

```
ctx = modbus_new_rtu(xxxxx);
if (!ctx) {
       fprintf(stderr, "things are b0rked: %s\n", modbus_strerror(errno);
}
```

mhdazeem, 2015-10-27:
Main Issue is I cannot see the error no. I am using docklight scripting for verbose. The only thing I get is "Connection Failed"

stephane, 2015-10-27:
@mhdazeem, the issue tracker is reserved for identified bugs.
For your problem, you must use the libmodbus mailing list with a detailed description (as already asked).

closed 2015-10-27 by stephane