stephane/libmodbus #294: AIX socket don't have MSG_DONTWAIT tag

poulacou, 2015-10-30:
MSG_DONTWAIT could be replaced by  MSG_NONBLOCK on AIX

** on Linux we have:
# grep MSG_DONTWAIT /usr/include/bits/socket.h
    MSG_DONTWAIT	= 0x40, /* Nonblocking IO.  */
#define MSG_DONTWAIT	MSG_DONTWAIT

** on AIX
# grep MSG_NONBLOCK /usr/include/sys/socket.h 
#define MSG_NONBLOCK    0x4000          /* nonblocking request */

poulacou, 2015-10-30:
I propose this change 
diff --git a/src/modbus-tcp.c b/src/modbus-tcp.c
index 00de11c..5c92743 100644
--- a/src/modbus-tcp.c
+++ b/src/modbus-tcp.c
@@ -440,9 +440,10 @@ static int _modbus_tcp_flush(modbus_t *ctx)
     do {
         /* Extract the garbage from the socket */
         char devnull[MODBUS_TCP_MAX_ADU_LENGTH];
-#ifndef OS_WIN32
-        rc = recv(ctx->s, devnull, MODBUS_TCP_MAX_ADU_LENGTH, MSG_DONTWAIT);
-#else
+#ifdef _AIX
+        rc = recv(ctx->s, devnull, MODBUS_TCP_MAX_ADU_LENGTH, MSG_NONBLOCK);
+
+#ifdef OS_WIN32
         /* On Win32, it's a bit more complicated to not wait */
         fd_set rset;
         struct timeval tv;
@@ -460,6 +461,9 @@ static int _modbus_tcp_flush(modbus_t *ctx)
             /* There is data to flush */
             rc = recv(ctx->s, devnull, MODBUS_TCP_MAX_ADU_LENGTH, 0);
         }
+#else
+        rc = recv(ctx->s, devnull, MODBUS_TCP_MAX_ADU_LENGTH, MSG_DONTWAIT);
+
 #endif
         if (rc > 0) {
             rc_sum += rc;

stephane, 2015-12-03:
Merged.

closed 2015-12-03 by stephane

poulacou, 2015-12-03:
Thanks ;-)

I can't test it  for the moment.
It looks ok from a functional point of view

stephane, 2015-12-04:
@poulacou I stay tuned, all feedback is always very appreciated.