stephane/libmodbus #302: Server on RaspberryPi (UART+GPIO) receive a <0> Byte at the beginning of each answer

BKedziora, 2015-12-09:
1.libmodbus version
- 3.1.2 - dhruvvyas90/libmodbus fork

2.OS/environment/architecture
- Raspberry Pi 2 with Raspbian Jessie
- RS485+GPIO switching with SN65HVD3085

3.libmodbus backend (TCP, RTU, IPv6)
- RTU

4.Modbus messages
TEST with /dev/ttyUSB0(with USB RS 485 dongle) - allways OK
**********************
pi@raspberrypi ~/SW-QT5/ModbusProjekt/ModbusProjekt/libmodbus-master/rpi-test $ sudo ./test
sec_to=0 usec_to=500000
Rpi RTU enabled.
BCM Pin set as 18.
RPI pin exported and pin direction configured successfully.
Opening /dev/ttyUSB0 at 9600 bauds (N, 8, 1)
[01][04][00][00][00][01][31][CA]
GPIO18 written : 1 successfully !
GPIO18 written : 0 successfully !
Waiting for a confirmation...
<01><04><02><03><28><B9><DE>
Date received is : 808
RPI BCM Pin Unexported successfully.

TEST with /dev/ttyAMA0 - allways <0> Byte... Error
**********************
pi@raspberrypi ~/libmodbus-master/rpi-test $ sudo ./test
sec_to=0 usec_to=500000
Rpi RTU enabled.
BCM Pin set as 18.
RPI pin exported and pin direction configured successfully.
Opening /dev/ttyAMA0 at 9600 bauds (N, 8, 1)
[01][04][00][00][00][01][31][CA]
GPIO18 written : 1 successfully !
GPIO18 written : 0 successfully !
Waiting for a confirmation...
ERROR Connection timed out: select
<00><01><04><02><03><25><78><1B>Bytes flushed (0)
Date received is : 0
RPI BCM Pin Unexported successfully.

pi@raspberrypi ~/libmodbus-master/rpi-test $ sudo ./test
sec_to=0 usec_to=500000
Rpi RTU enabled.
BCM Pin set as 18.
RPI pin exported and pin direction configured successfully.
Opening /dev/ttyAMA0 at 9600 bauds (N, 8, 1)
[01][04][00][00][00][01][31][CA]
GPIO18 written : 1 successfully !
GPIO18 written : 0 successfully !
Waiting for a confirmation...
ERROR Connection timed out: select
<00><01><04><02><03><28><B9><DE>Bytes flushed (0)
Date received is : 0
RPI BCM Pin Unexported successfully.

dhruvvyas90, 2015-12-09:
Did you turn off serial console on RPi which is turned on by default ? If not, use `sudo raspi-config`, option 8 (advanced options) --> choose A8 (serial) --->  choose option `no`. back --> back --> finish. Reboot and see if it works. Also, this is not original libmodbus/master issue and hence you should raise this issue on that particular fork or ask on libmodbus google group.

dhruvvyas90, 2015-12-09:
Make sure you're using correct GPIO numbering scheme (fork follows, BCM numbering) and correct GPIO number.

BKedziora, 2015-12-09:
1. Serial console is turned off
2. GPIO is correct
I  receive <00><01><04><02><03><28> instead of <01><04><02><03><28>.
               ******
When I interrupt the line to the slave I still receive a <0> Byte

dhruvvyas90, 2015-12-09:
I see, try using different baud rate if you can. I have observed this <0> problem before while using `ttyAMA0` at higher speeds (38400 bps and above) in older kernels, I believe it has something to do with RPi kernel, I've not investigated that issue further since it solved my problem by lowering my baud rate. Hope it helps.

dhruvvyas90, 2015-12-09:
I believe this is the issue you're facing `http://elinux.org/RPi_Serial_Connection#Glitch_when_opening_serial_port` Try querying more  with the open connection to see if it goes away with further more modbus queries.

BKedziora, 2015-12-09:
The problem exists both with 9600 and with 57600.
uname -a
Linux raspberrypi 4.1.7-v7+ #817 SMP PREEMPT Sat Sep 19 15:32:00 BST 2015 armv7l GNU/Linux

dhruvvyas90, 2015-12-09:
Or just flush buffer after opening the port.

BKedziora, 2015-12-09:
[image]

dhruvvyas90, 2015-12-09:
Did you try flushing the port after opening it ?

BKedziora, 2015-12-09:
in modbus-rtu.c

static int _set_GPIO_pin(modbus_t *ctx,int value)
{
  if(ctx->enable_rpi_rtu == 1)
  {
   ......
->  if(value == 0)
->  {
->      modbus_flush(ctx);
->  }
  }
  return 0;
}

added -> and the <0> Byte at beginning is away!

dhruvvyas90, 2015-12-09:
Instead of changing the source, you could have just used in your user program anywhere after `modbus_connect` function. Glad that your issue has been solved. You may close this issue now.

stephane, 2015-12-10:
@dhruvvyas90 Thank you for your help!

stephane, 2015-12-10:
I could be a good idea to add the modbus_flush() trick to https://github.com/stephane/libmodbus/wiki/Libmodbus-on-Raspberry-pi

dhruvvyas90, 2015-12-10:
@stephane: Let me modify the wiki post. You may close this issue since it has been solved.

BKedziora, 2015-12-11:
I changed "server" in "master" in order to guard against misunderstandings.

stephane, 2016-05-11:
@dhruvvyas90 thank you for your answers and the wiki. Closed.

closed 2016-05-11 by stephane

robetson1989, 2016-08-11:
hello happy to say hello. I can not compile the library in a raspberry pi who can help me? Greetings from Venezuela

pi@homecontrol:~/libmodbus/tests $ gcc random-test-server.c -o random-test-server pkg-config -libs -cflags libmodbus
gcc: error: pkg-config: No existe el fichero o el directorio
gcc: error: libmodbus: No existe el fichero o el directorio
gcc: error: unrecognized command line option '-cflags'

jose560, 2016-10-05:
hello, i´m trying to compile the modbus-rtu.c but there is a fatal error that says: "no such file or directory #include<config.h>"  can anyone help me ? Thanks

dannychris37, 2021-02-17:
Hi, does this problem persist in the current version? The solution to modify modbus-rtu.c no longer applies since the code in the post no longer exists.