stephane/libmodbus #303: Server on RaspberryPi(UART+GPIO) - the GPIO will be sometimes to late switchted off

BKedziora, 2015-12-11:
1.libmodbus version
- 3.1.2 - dhruvvyas90/libmodbus fork

2.OS/environment/architecture
- Raspberry Pi 2 with Raspbian Jessie
- RS485+GPIO switching with SN65HVD3085

3.libmodbus backend (TCP, RTU, IPv6)
- RTU

4.issue description
- the GPIO will be sometimes to late switchted off
	see oscillogram pcture

BKedziora, 2015-12-11:
[image]

BKedziora, 2015-12-11:
I changed "server" in "master" in order to guard against misunderstandings.

dhruvvyas90, 2015-12-12:
It could be because of additional flush you added in modbus-rtu file when GPIO goes low. I don't know time taken by that function but I suspect that if it's in few ms, it could delay GPIO going low and hence hurt the overall transmission. You can run time test specifically for modbus_flush function as mentioned in this link. http://stackoverflow.com/questions/5248915/execution-time-of-c-program Hope it help.

stephane, 2016-02-17:
@BKedziora do your read the comment of @dhruvvyas90?

stephane, 2016-05-11:
No news since dec. 2015. I would prefer to have answer from @BKedziora but the list of issues needs a spring cleaning.

closed 2016-05-11 by stephane