stephane/libmodbus #304: CRC calculation issue

jutshanahan, 2015-12-16:
I'm using a LinuxCNC plugin (aka HAL component) called vfdb_vfd to control a spindle via RS485 Modbus RTU.

When the component initialize, I observe a strange anomaly with respect to the CRC calculation.  For the bytes received below, the CRC shown at the end, 94FC is in fact correct.  But, libmodbus thinks it's B0E7.  If you append 0x00 to the beginning of this byte array, you will get the CRC that libmodbus calculates, B0E7.  

	halcmd: [01][03][01][00][00][01][85][F6]
	Waiting for a confirmation...
	<00><01><03><02><6D><60><94><FC>
	ERROR CRC received 94FC != CRC calculated B0E7

Note; this was used to calculate the CRC:  http://www.lammertbies.nl/comm/info/crc-calculation.html
Input:        0103026D60        Output:  0xFC94  CRC-16 (Modbus)
Input:    000103026D60        Output:  0xE7B0   CRC-16 (Modbus)

dhruvvyas90, 2015-12-16:
I believe your RTU response should be something like <01><03>... so the starting <00> could be the problem. You may confirm response on some kind of simulator or something. In that case, you may use `modbus_flush(ctx)` after `modbus_connect` and give it a shot.

jutshanahan, 2015-12-20:
The <00> at the start resulted from improper termination on the bus.  I had 120 ohm at each end of the bus, which at first seems like it's fine.  However, when the bus is not driven, aka neither device is transmitting, the voltage across the termination resistors is nearly zero volts.  This is outside the logic voltage thresholds for the RS485 transceiver on my FTDI usb dongle, which is:  Logic 1:  >200mV.  Logic 0:  < -200mV.  The solution was to termination the bus in a "fail safe" manner with a 560 ohm resistor from Data+ to 5V and another 560 ohm resistor from Data- to ground.

dhruvvyas90, 2015-12-20:
Good to know that issue has been solved. You may close this issue now.

closed 2015-12-20 by jutshanahan