stephane/libmodbus #307: modbus_new_tcp() doesn't appear to handle NULL for the _ip address argument

jfieldhouse, 2016-01-28:


stephane, 2016-01-29:
IP address is checked in master:
https://github.com/stephane/libmodbus/blob/master/src/modbus-tcp.c#L811

Which version did you use?

jfieldhouse, 2016-01-29:
I'm implementing a Modbus slave

jfieldhouse, 2016-01-29:
I believe i downloaded v3.0.6 and the source code for modbus-tcp. is different than what you are showing:

modbus_t* modbus_new_tcp(const char *ip, int port)
{
    modbus_t *ctx;
    modbus_tcp_t *ctx_tcp;
    size_t dest_size;
    size_t ret_size;

#if defined(OS_BSD)
    /* MSG_NOSIGNAL is unsupported on *BSD so we install an ignore
       handler for SIGPIPE. */
    struct sigaction sa;

    sa.sa_handler = SIG_IGN;
    if (sigaction(SIGPIPE, &sa, NULL) < 0) {
        /* The debug flag can't be set here... */
        fprintf(stderr, "Coud not install SIGPIPE handler.\n");
        return NULL;
    }
#endif

    ctx = (modbus_t *) malloc(sizeof(modbus_t));
    _modbus_init_common(ctx);

    /* Could be changed after to reach a remote serial Modbus device */
    ctx->slave = MODBUS_TCP_SLAVE;

    ctx->backend = &(_modbus_tcp_backend);

    ctx->backend_data = (modbus_tcp_t *) malloc(sizeof(modbus_tcp_t));
    ctx_tcp = (modbus_tcp_t *)ctx->backend_data;

    dest_size = sizeof(char) * 16;
    ret_size = strlcpy(ctx_tcp->ip, ip, dest_size);
    if (ret_size == 0) {
        fprintf(stderr, "The IP string is empty\n");
        modbus_free(ctx);
        errno = EINVAL;
        return NULL;
    }

    if (ret_size >= dest_size) {
        fprintf(stderr, "The IP string has been truncated\n");
        modbus_free(ctx);
        errno = EINVAL;
        return NULL;
    }

    ctx_tcp->port = port;

    return ctx;
}

jfieldhouse, 2016-01-29:
It looks like you've only implemented that ability in the latest version of the library. Is that correct?

jfieldhouse, 2016-01-29:
I see now that I was referring to the documentation for the latest code branch, but using v3.0.6 source code. Hence the disconnect I was seeing. I'll update to 3.1.2.
Thanks.

stephane, 2016-01-29:
Don't forget to close the issue, please.

closed 2016-01-29 by jfieldhouse