stephane/libmodbus #308: Bad File Descriptor

AndreySc, 2016-02-03:
Hello,
I am writing a program in C for a temperature controller (Eurotherm 2704 connected via RS232). 
I use Linux (opensuse) and I write the program with Codeblocks (gcc).
When I run the code I get the error message "Bad File Descriptor" 
( error code [01][03][00][18][00][01][04][0D] ) ( process returned 255 (0xFF) )
The strange thing is, that I can read and write registers with qmodbus 0.2, which is based on libmodbus. So communication is possible between the Pc and the controller. Any tips or ideas where the problem might be?

thanks
Andreas

code:
int main()
{

modbus_t *ctx;
uint16_t tab_reg[64];
int rc;
int i;

ctx = modbus_new_rtu("/dev/ttyS0", 9600,'N', 8, 1);
if (ctx == NULL) {
    fprintf(stderr, "Unable to create the libmodbus context\n");
    return -1;
}

modbus_set_debug(ctx, TRUE);

rc = modbus_rtu_set_serial_mode(ctx, MODBUS_RTU_RS232);
if (rc == -1) {
    fprintf(stderr, "serial mode failed: %s\n", modbus_strerror(errno));
    modbus_free(ctx);
    return -1;
}

rc = modbus_set_slave(ctx ,1);
if (rc == -1) {
    fprintf(stderr, "rtu failed: %s\n", modbus_strerror(errno));
    modbus_free(ctx);
    return -1;
}

rc = modbus_read_registers(ctx, 24, 1, tab_reg);
if (rc == -1) {
    fprintf(stderr, "%s\n", modbus_strerror(errno));
    return -1;
}

for (i=0; i < rc; i++) {
    printf("reg[%d]=%d (0x%X)\n", i, tab_reg[i], tab_reg[i]);
}

modbus_close(ctx);
modbus_free(ctx);
return 0;
}

stephane, 2016-02-03:
Are you sure about device name and permissions?

AndreySc, 2016-02-04:
I checked name and permissions.
When I use  the command "dmesg | grep tty", I get two answers:
[tty0] enabled
ttyS0 at I/O 0x3f8
ttyS4 at I/O 0xec98
I used "chmod 777" to give full permissions to both files in /dev/ , but I get still the error.
By the way, if I dont give ttyS0 permissions, qmodbus cant send/receive data and i get an error.
After giving full permissions to ttyS0 it runs.
So ttyS0 seems to be the right file.

Andreas

AndreySc, 2016-02-04:
one more thing. I installed first libmodbus 3.08 and than installed later 3.12 over the old installation.
I dont know if that can cause problems...

stephane, 2016-02-05:
Could you give the console output when you run the code?
Could you give me the output of `pkg-config --cflags --libs libmodbus`

AndreySc, 2016-02-08:
This is the console output when I run the program (compiled with gcc):

[01][03][00][18][00][01][04][0D]
ERROR Bad file descriptor
Bad file descriptor

This is the console output when I run the pkg-config as you described:

-I/usr/local/include/modbus -L/usr/local/lib -lmodbus 

Andreas

stephane, 2016-02-09:
I didn't catch it at first but you forget to call modbus_connect()

closed 2016-02-09 by stephane

stephane, 2016-02-09:
May be I could extend the error message with "Did you call modbus_connect()" ;)

Chadster766, 2016-02-09:
That's too funny!

stephane, 2016-02-09:
No seriously! That a common problem because the API isn't trivial on this point, I can't change the API but I can improve the help.

Chadster766, 2016-02-09:
Sorry I thought you were joking. The whole Modbus communication process starts with modbus_connect :smile:

xhtsansiro, 2023-08-10:
Does one have to invoke modbus_connect() every time in the loop of reading register information. In my case, I encounter the same error in the middle of reading process. In the beginning, everything goes fine, it keeps reading register, and to a point, it starts output Error: bad file descriptor