stephane/libmodbus #309: [MODBUS TCP] How to connect Multiple Clients with different Servers

kunjanpshah, 2016-02-06:
Dear All,

I have configured my device as **MODBUS TCP master/client**. And I want to connect multiple slave/server devices (different slave devices) using Modbus TCP. 

As per my understanding, in Modbus TCP protocol, we need to configure IP address of each slave and hence need to create and bind multiple sockets for connection establishment. 

So, could you please suggest me how to proceed with libmodbus library to connect multiple slaves on Modbus TCP.

Regards,
Kunjan Shah

stephane, 2016-03-07:
An example is https://github.com/stephane/libmodbus/blob/master/tests/bandwidth-server-many-up.c

closed 2016-03-07 by stephane

Ritesh236, 2019-10-31:
@stephane ,

We have also one requirement in which we want to use single server and multiple clients for same. So, We have checked with single server and two clients so far. 

Will it be possible to check with more clients like 4 to 5 with single server with this Modbus Library?

Regards,
Ritesh Prajapati

Jasper-Bai, 2023-01-10:
> @stephane ,
>
> We have also one requirement in which we want to use single server and multiple clients for same. So, We have checked with single server and two clients so far.
>
> Will it be possible to [...]

hi,
do you know how to assign different IP addr for various clients? for example, one PC contains multiple netcards.