stephane/libmodbus #311: RS485 DE signal control

lamazavr, 2016-02-09:
Hi, I have olinuxino-a10lime board with 4.1.3 kernel and libmodbus 3.0.5
I use: 
modbus_rtu_set_serial_mode(ctx, MODBUS_RTU_RS485);
but nothing happens. DE still on 0 level.
How can I control DE signal ?

itiserik, 2016-02-09:
Neither olilinuxino nor the allwinner a10 has native 485 mode (i.e. the at91sam9g45 has one look, at its datasheet)

Perhaps you can apply the patches in #156.

karlp, 2016-02-10:
some of modern kernels have started providing a way of providing a gpio to handle this from the kernel, but yeah, this is a very very unused part of the kernel, and the libmodbus hooks rely on the kernel doing the right thing.

lamazavr, 2016-02-11:
Hi, again! 
I recently updated my libmodbus to 3.1.2 version. On olinuxino-a10lime I used modbus_rtu_set_rts function and DE signal changes normally.

stephane, 2016-02-11:
Great, one more reason to release libmodbus 3.2 and to inform all Linux
distros.

lamazavr, 2016-02-12:
I tested library from main brunch and all is okay.

closed 2016-02-12 by lamazavr