stephane/libmodbus #316: modbus_set_serial_mode problem

cverster, 2016-03-16:
I am trying to set the serial mode to RS485.
When I run the modbus_get_serial_mode() function, it returns RS232.
When I then try to use the modbus_set_serial_mode(ctx, MODBUS_RTU_RS485); function, it returns the error: Inappropriate ioctl for device. I am using a USB to RS485 converter to communicate from my PC.

The real issues is that I am getting "Invalid CRC" errors when I try to read registers from the device, and I am assuming that the serial mode wrongly set is my culprit.

Help would be appreciated!

karlp, 2016-03-16:
If you're using a USB to rs485 converter, the kernel doesn't know that it's rs485, and doesn't need to enable any ioctrl settings for it.  those ioctls are only necessary when it needs to tell hardware to switch modes.  If you're getting CRC errors, but otherwise seem to be getting data and responses, check your parity settings.

stephane, 2016-03-22:
@cverster does it work now?

cverster, 2016-03-22:
Hi Stephane / Karl

I found that my problem was elsewhere and that it works fine after correcting that problem. Thank you very much and thank you for a great Modbus library!

Kind regards
Cornel Verster

closed 2016-03-22 by stephane