stephane/libmodbus #317: Message length not corresponding to the computed length

mmanoharan83, 2016-03-21:
When I'm connecting to device with modbus tcp (12 bit input) getting the below error . How to resolve the ERROR.

Logs:
==========
Connecting to XX.YY.ZZ.AA:502
[00][01][00][00][00][06][FF][05][00][00][FF][00]
Waiting for a confirmation...
<00><01><00><00><00><04><FF><30><31>
Message length not corresponding to the computed length (9 != 12)
ERROR modbus_write_bit (-1)
Address = 0, value = 1
[00][02][00][00][00][0B][FF][0F][00][00][00][20][04][8B][29][ED][E7]
Waiting for a confirmation...
<0D><00><02><00><00><00><04><FF><30>
Invalid transaction ID received 0xD00 (not 0x2)
ERROR modbus_write_bits (-1)
Address = 0, nb = 32
[00][03][00][00][00][06][FF][06][00][00][D7][15]
Waiting for a confirmation...
ERROR Invalid data: too many data
<31><0D><00><03><00><00><00><04><FF>ERROR modbus_write_register (-1)
Address = 0, value = 55061 (0xD715)
[00][04][00][00][00][47][FF][10][00][00][00][20][40][D7][15][64][F5][C8][78][CC][65][E9][60][32][92][55][D0][C4][A9][47][1B][8D][D0][7A][36][A0][FD][5D][62][83][6D][F3][C4][EA][8A][A2][BD][B7][A0][24][3F][9B][61][04][2C][3E][2D][23][21][CD][DD][28][1B][66][A3][21][39][1B][DA][FF][B8][37][DF][83][4F][D6][CF]
Waiting for a confirmation...
<30><31><0D><00><04><00><00><00><04><FF><30><31><0D>
Invalid transaction ID received 0x3031 (not 0x4)
ERROR modbus_write_registers (-1)
Address = 0, nb = 32
[00][05][00][00][00][4B][FF][17][00][00][00][20][00][00][00][20][40][28][EA][9B][0A][37][87][33][9A][16][9F][CD][6D][AA][2F][3B][56][B8][E4][72][2F][85][C9][5F][02][A2][9D][7C][92][0C][3B][15][75][5D][42][48][5F][DB][C0][64][9E][FB][D3][C1][D2][DC][DE][32][22][D7][E4][99][5C][DE][C6][E4][25][00][47][C8][20][7C][B0][29][30]
Waiting for a confirmation...
<00><05><00><00><00><04><FF><30><31>
Message length not corresponding to the computed length (9 != 73)
ERROR modbus_read_and_write_registers (-1)
Address = 0, nb = 32
[00][06][00][00][00][06][FF][05][00][01][FF][00]
Waiting for a confirmation...
<0D><00><06><00><00><00><04><FF><30>
Invalid transaction ID received 0xD00 (not 0x6)
ERROR modbus_write_bit (-1)
Address = 1, value = 1
[00][07][00][00][00][0B][FF][0F][00][01][00][1F][04][0D][10][4D][15]
Waiting for a confirmation...
ERROR Invalid data: too many data
<31><0D><00><07><00><00><00><04><FF>ERROR modbus_write_bits (-1)
Address = 1, nb = 31
[00][08][00][00][00][06][FF][06][00][01][9C][D5]

stephane, 2016-03-22:
I think the error is in the remote device (wrong response)

closed 2016-05-11 by stephane

mmanoharan83, 2016-05-13:
This is repeatedly occurs when receiving data from e-DAM 9050A module. This works on Modbus-TCP

RadonicA, 2023-03-22:
Hi, Ihave similary problem like @mmanoharan83 .
//MASTER
#include <stdio.h>
#include <stdlib.h>
#include <modbus.h>
#include <errno.h>

int main(int argc, char *argv[]) {
modbus_t *ctx;
uint32_t tab_reg;
int rc,i;

ctx = modbus_new_tcp("192.168.100.102", 502);
modbus_connect(ctx);

rc = modbus_read_registers(ctx, 0, 32, tab_reg);

if (rc == 10) {
    printf("Read registers: ");
   // for (i=0; i<10; i++) {
        printf("%f ", tab_reg);
    }
    printf("\n");
   // printf("%f\n", *(float*)tab_reg);  // added because my server had a float value in the first two registers (32 bits)
} else {
    fprintf(stderr, "Failed to read registers: %s\n", modbus_strerror(errno));
}

modbus_close(ctx);
modbus_free(ctx);
return 0;
}

//SLAVE
#include <stdio.h>
#include <stdlib.h>
#include <modbus.h>
#include <errno.h>

/* Temperatura od Raspberry Pi */

float get_cpu_temp() {
   FILE *fp;
   float temp;
    fp = popen("/usr/bin/vcgencmd measure_temp | awk -F '=' '{print $2}'", "r");
   if (fp == NULL) {
      printf("Failed to get temperature\n" );
      exit(1);
 }
   fscanf(fp, "%f", &temp);
    pclose(fp);
    return temp;
}
/* Kraj */

int main(int argc, char *argv[]) {

int temp=(int)get_cpu_temp();
printf("%d\n",temp);
// dobro ovo gore
modbus_t *ctx;
//modbus_mapping_t *mb_mapping;
uint16_t tab_reg[20];  
int socket;

printf("starting modbus server\n");
ctx = modbus_new_tcp(NULL, 502);
modbus_set_debug(ctx, 1);
mb_mapping = modbus_mapping_new(0, 0, 20, 0); //mapira jedan registar
modbus_set_float_dcba(temp, mb_mapping->tab_registers);
socket = modbus_tcp_listen(ctx, 1);
modbus_tcp_accept(ctx, &socket);

for (;;) {
    uint8_t query[MODBUS_TCP_MAX_ADU_LENGTH];
    int rc;

    rc = modbus_receive(ctx, query);

    if (rc > 0) {
//modbus_write_register(ctx,0,temp);
        modbus_reply(ctx, query, rc, mb_mapping);
    } else if (rc == -1) {
        break;
    }
}

printf("Quit the loop: %s\n", modbus_strerror(errno));

modbus_mapping_free(mb_mapping);
modbus_close(ctx);
modbus_free(ctx);

return 0;
}