stephane/libmodbus #320: modbus_new_rtu() 5,6,7 data bits

bmr006, 2016-04-11:
One of the parameters of modbus_new_rtu() - is the number of data bits.
It can be 5,6,7,8.
In my opinion for Modbus RTU protocols the number f data bits should be 8.
The function should not allow 5,6,7 data bits

karlp, 2016-04-11:
Unfortunately, not every device out there implements the standards properly.  While users should be discouraged from using odd bit sizes, maintaining the ability to talk to devices using non-standard line formats is more helpful than simply banning it.

bmr006, 2016-04-11:
Since this library is for Modbus & not a general purpose RS232 UART library, - it makes sense not to allow 5/6/7 data bits for the Modbus binary protocol.
I agree that for general purpose RS232 communication library this option should be given.

itiserik, 2016-04-11:
Well, if we had modbus ascii support in this lib then 7 bits would be possible.

stephane, 2016-05-17:
I'm not willing to add restrictions on these values, the related changes aren't worth the pain (new user messages, documentation update, tests).

closed 2016-05-17 by stephane