stephane/libmodbus #328: Not able to test

sachinkp21587, 2016-05-17:
i am new to linux over rpi. i get following when i issue command "make check"
`pi@raspberrypi ~/libmodbus $ make check
Making check in src
make[1]: Nothing to be done for 'check'.
Making check in doc
make[1]: Nothing to be done for 'check'.
Making check in tests
make  check-TESTS
Starting server
Starting client
unit-test-server: no process found
FAIL: ./unit-tests.sh
=============================================================
1 of 1 test failed
Please report to https://github.com/stephane/libmodbus/issues
=============================================================
Makefile:492: recipe for target 'check-TESTS' failed
make[2]: *** [check-TESTS] Error 1
Makefile:615: recipe for target 'check-am' failed
make[1]: *** [check-am] Error 2
Makefile:426: recipe for target 'check-recursive' failed
make: *** [check-recursive] Error 1`

Kindly guide.

stephane, 2016-05-17:
I don't know why you don't have `unit-test-server`(unit-test-server: no process found) neither the release you use, may be it's related to your Raspberry Pi env...

stephane, 2016-05-17:
Could you use the mailing list to further talk about your issue, please?

closed 2016-05-17 by stephane

sachinkp21587, 2016-05-19:
Hi,

That fail problem got resolved when i gave all permissions to the folder.
I have one other big problem which is not related ti this may be bt i am
new to linux so just want to ask u.
I downloaded and compiled libmodbus using steps on
https://github.com/stephane/libmodbus/wiki/Libmodbus-on-Raspberry-pi
Now when ever i restart my RPi the libmodbus folder gets deleted every time.

Kindly help.

Thanks,
Sachin kumar pawar