stephane/libmodbus #332: problem with polling multiple slave through one slave coordinator

damianoct, 2016-06-01:
Hi,
I communicate with libmodbus (debian package) with one coordinator/gateway connected via serial port.
Through it I can access to other slave data connected to the coordinator. I want to run a cyclic polling sending many message to multiple slave in one cycle.
For example, in a cycle I set the slave address in the context with modbus_set_slave() function and then send a read request and so on until all slaves are reached.

This system seems very slowly and often I receive an error of read, because, (I think) I change the slave address in the context without wait any time.

It is possible to poll a large number of slave connected to a single coordinator in a proper way?

Thanks in advance.

stephane, 2016-06-02:
The modbus_set_slave() call is very fast(https://github.com/stephane/libmodbus/blob/master/src/modbus-rtu.c#L92

the problem is certainly on your serial link (try to increase bauds if you can).

I think it's the proper way to read data from multiple slaves.

closed 2016-06-02 by stephane

chaikonen5, 2019-05-27:
Can you call modbus_set_slave() without Close connection? To pool diferent slave with the same Bus Pointer*??

Thanks,

stephane, 2019-05-27:
Yes (on same serial line)!