stephane/libmodbus #333: can we remove the listening socket nuke at modbus-tcp.c:644

codersmurf, 2016-06-01:
### libmodbus version
all

### Operating system
all

### Description of the Modbus network (server, client, links, etc)
server

### Expected behavior
should not nuke the socket sent in to the accept

### Actual behavior
nukes the socket sent into the accept

### Steps to reproduce the behavior (commands or source code)
create a non-blocking listening socket and use it in the accept, the socket is obliterated

### libmodbus output with debug mode enabled
the lib doesn't mind so no debug errors

Thanks,

Jake

stephane, 2016-06-02:
There is nothing at this line:
https://github.com/stephane/libmodbus/blob/master/src/modbus-tcp.c#L644

Which version do you use?

codersmurf, 2016-06-02:
Sorry about that, it starts on line 663, if the socket accept comes
back with a bad socket it nukes the listening socket.

Thanks,

Jake

codersmurf, 2016-07-18:
Were you ever able to see what I was talking about?

Thanks,

stephane, 2016-07-18:
https://github.com/stephane/libmodbus/blob/master/src/modbus-tcp.c#L663

If the accept's socket is invalid, the listening socket is closed.
Why do you want a different behaviour?

codersmurf, 2016-07-18:
So I've put the socket in non-blocking and if there is an error
because there is no data the socket gets closed out from underneath
me.

Thanks,

Jake

stephane, 2016-07-18:
I see

stephane, 2016-07-19:
Could you provide a very simple piece of code which put the socket in non-blocking and listen?
This way I'll be able to look at the error code (errno) and write a unit test with that.

mhei, 2016-07-19:
When I understand @codersmurf correctly, we do not need a unit test for this, just simply remove `close(*s)` and `*s = -1`. In other words, the library is given a library-external acquired resource and when this is not usable and/or not in correct state, then the library should not touch it further. This should be left to the application to deal with.
An error within the library is already signaled by return code of the function. Application should check for this.
Also the documentation does not state that the parameter is mangled in case of error.

stephane, 2016-07-19:
Oh I didn't awake this morning... the socket is created by `modbus_tcp_socket()` but any valid socket could be given to `modbus_tcp_accept()` and this socket isn't stored in modbus context so you're right, the resource management belongs to the application.

https://github.com/stephane/libmodbus/blob/master/src/modbus-tcp.c#L477

closed 2016-07-19 by stephane

codersmurf, 2016-07-19:
Did u still want some code that puts the socket in nonblocking?

commit 00e068eb (2016-07-19): tcp_modbus_accept mustn't close socket on error (closes #333)

The socket management belongs to the application.

The return code was missing in modbus_tcp_pi_accept().