stephane/libmodbus #334: Memory Leak in _modbus_tcp_flush.

preetamj84, 2016-06-02:
### libmodbus version

### Operating system

### Description of the Modbus network (server, client, links, etc)

### Expected behavior

### Actual behavior

### Steps to reproduce the behavior (commands or source code)

### libmodbus output with debug mode enabled

stephane, 2016-06-02:
In your report, you have indicated **all** modbus versions but this message (Invalid TID received) doesn't exist in master branch. Which version do you use?

BTW a message with '0 bytes freed' doesn't mean there is a leak in the program.

preetamj84, 2016-06-03:
Lib Modbus Version is 3.0.6

And the statement appears in the following code in modbus_tcp.c
int _modbus_tcp_pre_check_confirmation(modbus_t *ctx, const uint8_t *req,
                                       const uint8_t *rsp, int rsp_length)
{
    /* Check TID */
    if (req[0] != rsp[0] || req[1] != rsp[1]) {
        if (ctx->debug) {
            fprintf(stderr, "Invalid TID received 0x%X (not 0x%X)\n",
                    (rsp[0] << 8) + rsp[1], (req[0] << 8) + req[1]);
        }
		errno = LT_INVALID_DATA;
        return -1;
    } else {
        return 0;
    }
}

This routine is called from the function check_confirmation in modbus.c

>> BTW a message with '0 bytes freed' doesn't mean there is a leak in the program.
In Task manager, the memory shows as increased after this point.

stephane, 2016-09-07:
Before further investigation, do you have the same behaviour with libmodbus v3.1.4 (or master) ?

studioraga, 2017-06-20:
I am seeing the same error while using ./unit-test-client tcp 

Output that appears is "" Invalid TID received.

This is seen with stable version 3.0.6 and master v3.1.4) ...