stephane/libmodbus #335: Missing file in git clone

mvkdeepak, 2016-06-03:
### libmodbus version

### Operating system

### Description of the Modbus network (server, client, links, etc)

### Expected behavior

### Actual behavior

### Steps to reproduce the behavior (commands or source code)

### libmodbus output with debug mode enabled

mvkdeepak, 2016-06-03:
I have git cloned the lib modbus v3.1.4 and tried to install on my desktop. But I am unable to as there are some files missing compared to the source tar which I subsequently downloaded.
The configure and build-aux is missing and I unable to proceed further.

Kindly help

stephane, 2016-06-03:
Did you read the README?

closed 2016-06-03 by stephane

mvkdeepak, 2016-06-09:
Dear Stephane I read it again and I clearly missed to execute the autogen.sh first. Sorry for the trouble and keep up the great work !!!