stephane/libmodbus #337: openwrt and libmodbus

heroseller17, 2016-06-14:
Hello
i have a question...i want to impliment modbus tcp/ip on soc...i RT5350 powefull chip and it support openwrt.
the question is can i use libmodbus on openwrt?
thanks

thatcadguy, 2016-06-14:
Hi, not a dev for libmodbus but as long as it is compatible with big-endian platforms (MIPS in this case), should be good to go.

mhei, 2016-06-14:
Yes, you can. Libmodbus is packaged for OpenWrt/LEDE, see https://github.com/openwrt/packages/tree/master/libs/libmodbus. However, please close this ticket, it's not an issue.

heroseller17, 2016-06-14:
Hi,
@mhei im a little confused! i was searching and i know about libmodbus thats interesting library ....
but i really dont know so much about openwrt....i want to run modbus tcp/ip slave wifi..is it possible with openwrt?
excause me if my question is so simple.

mhei, 2016-06-14:
libmodbus is only a library, you will need to write your own daemon. but in principle, it is possible with openwrt.

heroseller17, 2016-06-15:
Hi
@mhei can i use libmodbus examples on openwrt? or i shoud customize libmodbus for my self?

mhei, 2016-06-15:
You could use the examples, but they are not packaged by default.

heroseller17, 2016-06-15:
Aha...Thanks alot so i buy Hardware (RT5350) asap and show my results here:)
@mhei  Thanks alot.

mhei, 2016-06-15:
Please don't forget to close this issue. Use mailing list for further need/reports...

heroseller17, 2016-06-15:
Hi
@mhei on Which board or processor you make this library(libmodbus)? 
is it possible to make it on wrtnode or other board that support openwrt?

karlp, 2016-06-15:
just close this.  libmodbus is packaged for _all_ platforms that openwrt/lede supports.  if openwrt works on the board, libmodbus works on the board.

closed 2016-06-16 by heroseller17