stephane/libmodbus #350: MinGW : implicit declaration of function 'getaddrinfo' and 'freeaddrinfo'

SESA258801, 2016-08-10:
### libmodbus version
3.0.6
### Operating system
Win 7 Pro 64bits
### Description of the Modbus network (server, client, links, etc)
Build with MinGW
### Expected behavior
Build successful
### Actual behavior
Build stop with error:
$ make
make  all-recursive
make[1]: Entering directory `/c/Users/SESA258801/Downloads/libmodbus-3.0.6'
Making all in src
make[2]: Entering directory `/c/Users/SESA258801/Downloads/libmodbus-3.0.6/src'
  CC       modbus.lo
  CC       modbus-data.lo
  CC       modbus-rtu.lo
  CC       modbus-tcp.lo
modbus-tcp.c: In function '_modbus_tcp_pi_connect':
modbus-tcp.c:311:5: error: implicit declaration of function 'getaddrinfo' [-Werror=implicit-function-declaration]
     rc = getaddrinfo(ctx_tcp_pi->node, ctx_tcp_pi->service,
     ^
modbus-tcp.c:336:5: error: implicit declaration of function 'freeaddrinfo' [-Werror=implicit-function-declaration]
     freeaddrinfo(ai_list);
     ^
cc1.exe: all warnings being treated as errors
make[2]: *** [modbus-tcp.lo] Error 1
make[2]: Leaving directory `/c/Users/SESA258801/Downloads/libmodbus-3.0.6/src'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/c/Users/SESA258801/Downloads/libmodbus-3.0.6'
make: *** [all] Error 2

For information:
w32api.h is already included from stdio.h
In file included from c:\mingw\include\w32api.h:59:0,
                 from c:\mingw\include\_mingw.h:73,
                 from c:\mingw\include\stdio.h:56,
                 from modbus-tcp.c:19:
c:\mingw\include\sdkddkver.h:184:0: note: this is the location of the previous definition
 #  define _WIN32_WINNT _WIN32_WINNT_WIN2K

I changed the code as follows in modbus-tcp.c and the build went well:
(I moved the #if defined(_WIN32) at the very beginning of the file)
...
#if defined(_WIN32)
# define OS_WIN32
/* ws2_32.dll has getaddrinfo and freeaddrinfo on Windows XP and later.
 * minwg32 headers check WINVER before allowing the use of these */
# ifndef WINVER
# define WINVER 0x0501
# endif
#include <stdio.h>
...
### Steps to reproduce the behavior (commands or source code)
Launch build
### libmodbus output with debug mode enabled

stephane, 2016-09-09:
@SESA258801  could check my patch, please?

SESA258801, 2016-09-09:
Build went well. Thx

stephane, 2016-09-12:
Thank you for the feedback. Merged in v3.0.X and master.

commit 692c43e3 (2016-09-09): Move WINVER definition before other includes (closes #350)
commit 419ef441 (2016-09-09): Move WINVER definition before other includes (closes #350)
closed 2016-09-12 by stephane