stephane/libmodbus #353: make error on Fedora9

stuyou, 2016-08-26:
### libmodbus version
 libmodbus-3.0.6.tar.gz (2013-10-06)
### Operating system
Fedora9
### Description of the Modbus network (server, client, links, etc)

### Expected behavior

### Actual behavior

### Steps to reproduce the behavior (commands or source code)
./configure
make
### libmodbus output with debug mode enabled
[root@localhost libmodbus-3.0.6]# make
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/sh /opt/libmodbus/libmodbus-3.0.6/missing aclocal-1.14 -I m4
/opt/libmodbus/libmodbus-3.0.6/missing: line 81: aclocal-1.14: command not found
WARNING: 'aclocal-1.14' is missing on your system.
         You should only need it if you modified 'acinclude.m4' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'aclocal' program is part of the GNU Automake package:
         <http://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <http://www.gnu.org/software/autoconf>
         <http://www.gnu.org/software/m4/>
         <http://www.perl.org/>
make: *** [aclocal.m4] Error 127

stephane, 2016-09-05:
Could you use libmodbus package provided by your distro?

stephane, 2016-09-09:
or install the proper version of aclocal...

closed 2016-09-09 by stephane