stephane/libmodbus #354: modbus_close() fails after modbus_connect() on a disconnected port

normads, 2016-09-04:
### libmodbus version
3.1.4
### Operating system
Linux version 4.8.0-rc1-ga0cba21
### Description of the Modbus network (server, client, links, etc)
Modbus RTU client
### Expected behavior
If modbus_connect() is called on a disconnected socket, ctx->s should not be replaced by -1, as it removes the possibility of closing the connection(modbus_close) later.
### Actual behavior
modbus_connect() called on a disconnected socket replaces ctx->s with -1 and the socket cannot be closed afterwards and the kernel will therefore not reuse the minor number. If the serial device gets disconnected frequently the kernel will eventually run out of minor numbers.  
### Steps to reproduce the behavior (commands or source code)

### libmodbus output with debug mode enabled

stephane, 2016-09-09:
I don't see where you see this problem.
Could you me the github link (file and line) of this problem in master?

normads, 2016-09-09:
> [https://github.com/stephane/libmodbus/blob/master/src/modbus-rtu.c](line 591)

modbus_connect() -- connects to a connected device
reads registers
device disconnects
read register fails
check if connection is available with if(modbus_connect(ctx) != -1)
ctx->s is replaced with -1 because the device is now disconnected
now the handle can never be closed because the file descriptor for the earlier connection has been replaced with -1
so the kernel will continue to issue a new minor number until the minor numbers run out

This is not a bug per say. Everything is fine if you call modbus_close() everytime prior calling modbus_connect(). But this requirement is not emphasized in the documentation and the idea i got from the documentation is that you can use modbus_connect() to check for an available connection. 

The way i see it, this would be a more intuitive approach,
`int tmpfd = open(ctx_rtu->device, flags);
 if (tmpfd == -1) {
        if (ctx->debug) {
            fprintf(stderr, "ERROR Can't open the device %s (%s)\n",
                    ctx_rtu->device, strerror(errno));
        }
        return -1;
}
ctx->s = tmpfd;`

This way even if the device was disconnected at the time of calling modbus_connect ctx->s would still hold the last file descriptor. And modbus_close() can be called at a later time. As shown in the documentation.