stephane/libmodbus #355: file not recognized: File format not recognized

SvenKortekaas, 2016-09-05:
### libmodbus version
3.1.4

### Operating system
Linux Ubuntu
Eclipse Version: Mars.2 Release (4.5.2)
Build id: 20160218-0600

### Expected behavior
Building the source code with Eclipse.

### Target environment
Raspberry Pi 2

### Actual behavior
Compiling went well but the Linker gave me an error. Got this error:

`/usr/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/4.9.3/../../../../arm-linux-gnueabihf/lib/libmodbus.so: file not recognized: File format not recognized`

### Steps to reproduce the behavior (commands or source code)
My project in Eclipse. Did these steps:

`$ git clone git://github.com/stephane/libmodbus`

`autogen.sh`

`./configure --prefix=/usr/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf/arm-linux-gnueabihf --host=arm-linux-gnueabihf`

`make`

`make install`

Added the library to my Eclipse project. In the Linker -> Libraries I added modbus and in the Library search path I added /usr/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf/arm-linux-gnueabihf/include/modbus and /usr/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf/arm-linux-gnueabihf/include.

The compiling went great but the linker gave me this:

`Invoking: Cross ARM C++ Linker`

`arm-linux-gnueabihf-g++ -mcpu=cortex-a9 -march=armv7-a -mthumb -O0  -g -Xlinker --gc-sections -s -L/usr/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf/arm-linux-gnueabihf/include -L/usr/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf/arm-linux-gnueabihf/include/modbus -Wl,-Map,"<projectname>_debug.map" -o "<projectname>_debug"  ./src/utilities/log.o ./src/utilities/oad_file.o  ./src/service/bluetooth.o ./src/service/database.o ./src/service/network.o  ./src/modbus/modbus_config.o ./src/modbus/modbus_send.o ./src/modbus/modbus_sensors_up.o ./src/modbus/modbus_server.o ./src/modbus/modbus_thread.o ./src/modbus/modbus_up_control.o  ./src/domain/command.o ./src/domain/device.o ./src/domain/event.o ./src/domain/measurement.o ./src/domain/scan.o  ./src/control/control.o  ./src/main.o  /home/<name>/Workspace/libshared-glib.a /home/<name>/Workspace/libbluetooth-internal.a -lstdc++ -lbacnet -lffi -lmodbus -lgobject-2.0 -lglib-2.0 -lgio-2.0 -lgmodule-2.0 -lconfig++ -lz -lcurl -lsqlite3`

`/usr/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/4.9.3/../../../../arm-linux-gnueabihf/lib/libmodbus.so: file not recognized: File format not recognized`

`collect2: error: ld returned 1 exit status`

`makefile:62: recipe for target '<projectname>_debug' failed`

SvenKortekaas, 2016-09-05:
I found the problem.

Needed to comment out: AC_FUNC_MALLOC 

in configure.ac and run this command:

`./autogen.sh && ./configure --prefix=/usr/gcc-linaro-4.9-2014.11-x86_64_arm-linux-gnueabihf/arm-linux-gnueabihf --host=arm-linux-gnueabihf && sudo make clean && sudo make && sudo make install`

closed 2016-09-05 by SvenKortekaas

danghuybk, 2018-11-22:
I met this problem. I read your comment but i dont know where is configure.ac ? Can you help me? Thanks you.