stephane/libmodbus #364: RTU RS485 Library Issue

demonphoenix37, 2016-11-17:
## libmodbus version

libmodbus 3.1.4

## OS and/or distribution

Ubuntu 14.04.5 LTS (GNU/Linux 3.16.0-77-generic x86_64)

## Description

All requests to read registers from a device result in a time out because rts and serial_mode parameters can not be set correctly. 

Every instance of "if (ctx->backend->backend_type == _MODBUS_BACKEND_TYPE_RTU) { ... }" in modbus-rtu.c effectively prevents functions like modbus_rtu_set_rts and modbus_rtu_set_serial_mode from functioning properly since backend_type is never set anywhere.

##Solution / Work Around

This can be corrected by setting parameter backend_type to _MODBUS_BACKEND_TYPE_RTU in the modbus_new_rtu function. This will allow modbus rtu commands such as modbus_rtu_set_rts and modbus_rtu_set_serial_mode to function correctly.

ctx->backend->backend_type == _MODBUS_BACKEND_TYPE_RTU;

##Other Notes

I'm submitting this issue in hopes it gets visibility for those who've struggled to find a solution for their RTU RS485 woes. Keep up the good work Stephane I really appreciate it!

stephane, 2016-11-18:
The backend_type is set by:
https://github.com/stephane/libmodbus/blob/master/src/modbus-rtu.c#L1189

and pointer to that structure it set here:
https://github.com/stephane/libmodbus/blob/master/src/modbus-rtu.c#L1233

so I don't know why you aren't able to pass the condition.

demonphoenix37, 2016-11-18:
backend data might be getting set but backend_type appears to not be set
anywhere in modbus.c or modbus-rtu.c. The if conditions in the functions I
mentioned are specifically targeting backend_type.

stephane, 2016-11-19:
The type of structure _modbus_backend defines backend_type: https://github.com/stephane/libmodbus/blob/master/src/modbus-private.h#L68

then a structure called _modbus_rtu_backend is created and filled with the _MODBUS_BACKEND_TYPE_RTU;.

In doubt, you can printf() the value of the field in `modbus_rtu_set_serial_mode` or compare it with the value of the field in modbus-tcp.c but I'm sure this field is properly set.

Do you try to use `modbus_rtu_set_serial_mode` before `mobus_new_rtu`, btw I intend to deprecate this function because it's almost useless and very confusing (only works for very specific devices).

stephane, 2017-01-20:
No answer.

closed 2017-01-20 by stephane