stephane/libmodbus #369: modbus_rtu_set_custom_rts problem

gianlucadisimone, 2016-12-14:
---

## libmodbus version

libmodbus-3.4.1

## OS and/or distribution

OpenWRT CC compiled for mips

## Description

Hi used function modbus_rtu_set_custom_rts(), for manage trasmition direction using a gpio of my HW.

I insert this function in main modbus 
.....
 
     ctxRTU = modbus_new_rtu("/dev/ttyS0", 19200, 'N', 8, 1);
    modbus_connect(ctxRTU);
    modbus_rtu_set_custom_rts(ctxRTU, csetrts); 

 .....
 and definited 

void csetrts(modbus_t *ctx, int on){
    int gpio = 20;
    printf("valore di on=%d\n",on);
    if (on){
    	SetOut(gpio, 1);
    }
    else{
    	SetOut(gpio, 0);
    }
}

where function SetOut () set pin of my HW

but libmodbus never call csetrts for set my pin during polling, so i can't read answer to my polling

any ideas?

Gianluca

stephane, 2017-03-22:
Do you check the return value of `modbus_rtu_set_custom_rts` to ensure it's working as expected?
(0 means success).
The RTS mode must be set with modbus_rtu_set_rts() to allow the call of your custom function:
https://github.com/stephane/libmodbus/blob/master/src/modbus-rtu.c#L286

gianlucadisimone, 2017-05-20:
Hi, I check return of function...

`//    printf("open modbus RTU\n");
    ctxRTU = modbus_new_rtu("/dev/ttyS0", 900, 'N', 8, 1);
//	    ctxRTU = modbus_new_rtu("/dev/ttyUSB0", 9600, 'N', 8, 1);
   //ctxRTU = modbus_new_rtu(mb->PortModBus, mb->baud_rate, 'N', 8, 1);
    //modbus_set_slave(ctxRTU, 1);
    modbus_connect(ctxRTU);
    //modbus_set_debug(ctxRTU,1);

 //   rc1=modbus_rtu_get_serial_mode(ctxRTU);
//    printf("modbus_rtu_get_serial_mode: %d \n",rc1);
  //  rc1=modbus_rtu_set_serial_mode(ctxRTU, MODBUS);
//    printf("modbus_rtu_set_serial_mode: %d \n",rc1);
    res = modbus_rtu_set_rts(ctxRTU, MODBUS_RTU_RTS_UP);
    printf("modbus_rtu_set_rts: %d \n",res);
   // modbus_rtu_set_rts_delay(ctxRTU,1000);
   res=modbus_rtu_set_custom_rts(ctxRTU, csetrts()); //GPIO20
   printf("modbus_rtu_set_custom_rts: %d \n",res);`

.......
MDB TCP start loop
modbus_rtu_set_rts: 0 
modbus_rtu_set_custom_rts: 0 
start loop rtu
........
seem ok, but not work.
My bus  schematic is

RXTX connect to chip RS485, and mode of trasmition connect at gpio 20 of my CPU

Do you have a example code for my problem?

ths

zzeroo, 2017-09-27:
@gianlucadisimone What is the return value of `csetrts()`?
    
    printf("csetrts(): %d\n", csetrts());

Nykytas, 2025-03-25:
Hello!

I encountered the same issue. I called `modbus_rtu_set_custom_rts(externalCtx, controlRTS)`, but `controlRTS `was never triggered.

I added a call to` modbus_rtu_set_rts(externalCtx, MODBUS_RTU_RTS_UP);` right after` modbus_rtu_set_custom_rts(externalCtx, controlRTS)`, and `controlRTS `started being triggered.

Hope this helps someone!