stephane/libmodbus #373: Use poll() if available instead of select()

DcShrek, 2017-02-21:
---
## libmodbus version
  all

## OS and/or distribution
  some embedded linux like: uClinux,
  can also be present on other linux systems.

## Environment
  arm,
  can also be present on other architectures.

## Description
  select() in _modbus_tcp_select() in modbus_tcp.c hangs sometimes. Traced to:
 -FD_SETSIZE = 1024, all FD_*() macros operating according to that limit. 
 -accept() returning descriptor > 1023.

## Solve
 Change select() to poll() if available.
---

stephane, 2017-03-22:
It's a difficult change and I don't understand the goal?

closed 2017-05-05 by stephane

rakshitaSiemens, 2021-05-06:
we are also facing similar issue.
Is there a way to replace select with poll()