stephane/libmodbus #375: Help, step 4 of install fails

EC-123, 2017-03-07:
I was OK with the install and when I got to step 4, it failed:

configure: error: invalid variable name: `–prefix'

I ran grep:

pi@raspberrypi:~/libmodbus $ cat configure | grep -n ^prefix
765:prefix
817:prefix=NONE

It looks like there is something wrong somewhere in the ./configure?  

pi@raspberrypi:~/libmodbus $ pkg-conf --modversion libmodbus
-bash: pkg-conf: command not found

pi@raspberrypi:~/libmodbus $ sudo find / -type f -name libmodbus*
/home/pi/libmodbus/libmodbus.pc.in

I'm using a pi B+: Linux raspberrypi 4.4.34+ #930 Wed Nov 23 15:12:30 GMT 2016 armv6l GNU/Linux

zzeroo, 2017-03-08:
Please check your build system.

```bash
pi@raspberrypi:~/libmodbus $ pkg-conf --modversion libmodbus
-bash: pkg-conf: command not found
```

The `-bash: pkg-conf: command not found` indicates **your** build system is not correct (maybe `sudo apt install pkg-config` helps).

stefanofiorentino, 2017-03-08:
pkg-config is the right process to call, not pkg-conf.
consider to export the PKG_CONFIG_PATH with the path of your libmodbus
installation (if not in /usr/)

stephane, 2017-03-22:
Thanks for @EC-123 !

closed 2017-03-22 by stephane

symc87, 2018-08-13:
Hi, I got same error with @EC-123, how to solve this?

zzeroo, 2018-08-13:
@symc87 

The right command, as @stefanofiorentino mentioned, is 

```bash
pkg-config --modversion libmodbus
```

If you get something like this

```bash
# pkg-config --modversion libmodbus                                                                                                                                   08:53:18
Package libmodbus was not found in the pkg-config search path.
Perhaps you should add the directory containing `libmodbus.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libmodbus' found
```

Then you can try to find, system wide, your `libmodbus.pc`

```bash
sudo find / -type f -name libmodbus*
```

If you get the path to the lib, set the `PKG_CONFIG_PATH` like so

```bash
export PKG_CONFIG_PATH=$PKG_CONFIG_PATH:/path/to/libmodbus/ 
```

The command `pkg-config --modversion libmodbus` should now work.