stephane/libmodbus #376: TCP: behaviour for "unit identifier == 0" makes little sense.

magthe, 2017-04-06:
- *Which version of libmodbus are you using?* 3.1.4

 - *Which operating system are you using?* Linus (ArchLinux)

 - *Describe the exact steps which reproduce the problem*

## libmodbus version

3.1.4

## OS and/or distribution

Linux (ArchLinux)

## Environment

64 bit

## Description

I sent a message using ModbusTCP with a "unit identifier" of 0.

## Expected behaviour

A response to the message.

## Actual behaviour

No response is sent.

## Suspected place of bug

I think the bug is [here](https://github.com/stephane/libmodbus/blob/master/src/modbus.c#L991). I'm guessing, without having studied the code in detail, that the TCP request frame is scaled down to match an RTU frame. This means that the "unit identifier" has the same location as "slave address". The line I pointed to will drop responses for "slave address == 0".

This behaviour is, AFAIU, not according to standard. I can also not find any documentation of it.

mhei, 2017-04-07:
I would agree that this is a bug because as of my understanding the unit identifier can take any value. This is because a broadcast does not make sense in a TCP context - only when the Modbus server is a "Modbus TCP to RTU proxy"...
So could you try whether the patch in my branch fixes your issue?

magthe, 2017-07-03:
Yupp, looks good!

closed 2017-07-03 by stephane

stephane, 2017-07-03:
Thank you @mhei and @magthe

commit b1adc371 (2017-07-03): Fix modbus_reply for TCP when unit id == 0 (fixes #376)

According to the Modbus standard, the unit identifier is choosen by the
TCP client and can be any value:

"This field is used for intra-system routing purpose.  It is typically
used to communicate to a MODBUS+ or a MODBUS serial line slave through a
gateway between an Ethernet TCP-IP network and a MODBUS serial line.
This field is set by the MODBUS Client in the request and must be returned
with the same value in the response by the server."

So the current check in modbus_reply must be extended to check whether
we are in a RTU or TCP co [... cut]