stephane/libmodbus #377: modbus-data.c: fix for bswap_16 fallback

dreamflow, 2017-04-25:
when compiling the newest libmodbus master branch with
GCC 6.3.0 (MinGW-w64 MSYS) on Win7 32bit via the steps:

    autogen.sh
    ./configure
    make

make warns:

    modbus-data.c:54:4: warning: #warning "Fallback on C functions for bswap_16" [-Wcpp]
     #  warning "Fallback on C functions for bswap_16"

it can be resolved by adding in [src/modbus-data.c](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c) directly under the line 42

    // Since GCC >= 4.30, GCC provides __builtin_bswapXX() alternatives so we switch to them

these 2 new lines

    #    undef bswap_16
    #    define bswap_16 __builtin_bswap16

closed 2017-04-25 by stephane

stephane, 2017-04-25:
I should have tested to avoid a blind push...

dreamflow, 2017-04-26:
the TravisCI report in the revert commit https://github.com/stephane/libmodbus/commit/bf3004743cb5dcab6937946c7791770ed5d1f410 was very helpful to find the cause:
__builtin_bswap16 was in a later gcc version available than __builtin_bswap32 .
__builtin_bswap16 is [available](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=52624) in gcc since 4.8.0 .

therefore a possible solution would be in [src/modbus-data.c](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c) to change also [line 41](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c#L41) to

    # if GCC_VERSION > 480

all in all, if in [src/modbus-data.c](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c) the [lines 39-46](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c#L39-L46) would change to

    #if defined(__GNUC__)
    #  define GCC_VERSION (__GNUC__ * 100 + __GNUC_MINOR__ * 10)
    #  if GCC_VERSION > 480
    // GCC > 4.8.0 provides __builtin_bswapXX() alternatives so we switch to them
    #    undef bswap_16
    #    define bswap_16 __builtin_bswap16
    #    undef bswap_32
    #    define bswap_32 __builtin_bswap32
    #  endif
    #endif

then the __builtin_bswap16 could compile without the "fallback" warning in
MinGW-w64 windows builds with a gcc compiler > 4.8.0 .

today i've already made a live test on real modbus-RS485-hardware,
where i've used the new libmodbus function [modbus_get_float_badc](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c#L142)
with the above suggested patch. this [modbus_get_float_badc](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c#L142) function
uses [internaly](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c#L147) gcc's __builtin_bswap16 via bswap_16 .
the result of the hardwar-test is: it works fine.

stephane, 2017-04-26:
What do you think of my patch (still use bswap32 for older releases and >=)?

dreamflow, 2017-04-27:
your [patch](https://github.com/stephane/libmodbus/commit/005bcd6fbab124fa6043b0261aa7b64afc276e9a) is very good. i've test-compiled the [bswap_16-#377](https://github.com/stephane/libmodbus/tree/bswap_16-%23377)-branch
with the result: no compile-warnings/errors on Win7 32bit with
GCC 6.3.0 (MinGW-w64 MSYS).

stephane, 2017-04-27:
Merged. Thank you.

commit 1d3c6401 (2017-04-25): Add fallback to bswap_16 for gcc on Windows (closes #377)

Thanks to dreamflow for the issue.
commit bf300474 (2017-04-25): Revert "Add fallback to bswap_16 for gcc on Windows (closes #377)"

This reverts commit 1d3c6401f8186b54453ba64606758a90b6ec3fca.

TravisCI reports:
../src/.libs/libmodbus.so: undefined reference to `__builtin_bswap16'

with gcc 4.6.3