stephane/libmodbus #380: suppressing local echo on a 2-wires RS485

oliviercommelarbre, 2017-04-28:
Dear Stéphane, this is actually not a bug report but a proposal for improvement.

here is my use case with the blocking limitation i'm facing:
i've started using libmodbus on a raspberry Pi to control some quipment over an rs485 bus in a 2-wires configuration (that is in half-duplex configuration with RX = TX). 
The basic problem i face (and that i cannot workaround without a patch) is the fact that every modbus request command (for instance a modbus_read_registers, modbus_write_register, etc) will fail since the request sent on the line by the master gets automatically echoed onto the RX (since TX=RX), so that my UART (a USB to RS232/RS422/RS485 converter based on the widely used FDTI FT232 chipset) first returns the transmitted bytes in the read buffers as soon as the master attempts to read the slave reply message. Hence the reply is not recognised and the call systematically fails in error.

i beleive this is actually a general case preventing all potential users from using the library as soon as they will share a similar configuration, which i would think is quite widespread, that is: 
- a half-duplex modbus on serial 2-wires rs485, which is pretty standard
- using an UART without ADDC (Automatic Data Direction Control) capabilities (e.g. the Moxa adapters such as the UPort 1110 supports it but it is the only one as far as i could find...), which i think is also quite widespread nowadays by using USB to serial converters.

about resolving the pb:
resolving the pb is actually really trivial and i tested it successfully in my case with a quick patch. The principle is to read back the echoed bytes automatically after sending the master request and before attempting to read the slave reply. It works like a charm at least in my case. the change i applied lies only in _modbus_rtu_send. 
For a final solution i would propose to make this functionality activable upon user configuration (e.g. through a call similar to modbus_rtu_set/get_rts calls, for instance modbus_rtu_set/get_local_echo or something like that). If you think like me that it can be valuable, i can spend a bit more time on it and propose a patch via a pull request, please let me know! 
cheers, Olivier

karlp, 2017-04-29:
Your adapter is really broken.  There are use cases for local echo, and you're very correct, libmodbus doesn't support it.  But...  Even the venerable old ft232 has direction control built in, and enabled by default.  Your adapter is either wired badly, broken, or has had the ft232 onboard eeprom reprogrammed in such a way to break the functional default configuration.

If you're using an ft232 there's absolutely no need to be trying to use only a single pin uart. (which you _could_ do, as you're only using half duplex, and some hardware supports this)  You've got tx/tx on the ft232, and rx/tx on your rs485 transceiver.  If someone's wired them all together on your "rs485-usb adapter" throw it in the bin and buy another one.

oliviercommelarbre, 2017-04-29:
Hi Karl and thks a lot for your reply. I'm not sure to understand the second part of your msg. The slave hardware i want to control from the Pi (for info it is this one, sorry it is in italian: http://www.newtohm.it/wp-content/uploads/2016/10/DDC-mPID4DSP2.pdf) is only supporting half-duplex on a single twisted pair rs485 MTU so i have no choice there as i have only two wires on the slave side. Then i might have done something wrong in the wiring to the Pi USB adapter (the ft232 based UART) but i did a very simple thing, that is to wire RX=TX and connect the slave on that 2 wires bus, with the required resistor between +/- lines close to the slave side of the bus.

To be clear, the "rs485-usb adapter" is the FT232 based tranceiver i have on the Pi side (through one of the PI USB slots, it is precisely this adapter https://www.amazon.co.uk/d/Lens-Flash-Adapters-Converters/KALEA-INFORMATIQUE-Chipset-FT232-RS422-Allows-Material/B004BM3AQO). i have no idea what there is as rs485 tranceiver on the slave side to connect to the 2-wires. 
I have used the USB adapter right out of the box without any specific eeprom configuration or reconfiguration. am i doing something wrong?
thanks again

karlp, 2017-05-01:
It definitely sounds like you're doing something wrong.  Wire the A/B lines from your KALEA dongle to one of the ports of the DDC-mPID4, either M2-2A/2B or M2-1A/1B, whicever you've configured or enabled.

From the manual: http://www.kalea-informatique.com/mag/fr/product-2332688.htm  that's the "Signaux RS-485 : T/R+,T/R-.GND" lines on your dongle.  Then you plug your dongle into a usb port on your rpi and you're done.

The resistor is pretty optional at the speeds most modbus devices run at, I'd leave it off. (and if you're going to put one in, make sure it's the right one!)

oliviercommelarbre, 2017-05-26:
Hi Karl, thanks again and sorry for being so long to feedback i was away these last weeks.

what you've described for the wiring is exactly what i have, connected to M2-1A/1B of the DDC-mPID4 in my case. Note that M2-2A/2B are in fact exactly 2 wires (+ the GND off course), that is one combined +TX/RX and one combined -TX/RX.
Hence my precise wiring is: 
USB adapter on the Pi                DDC-mPID4
GND <-------------------> GND
T+   <-+-----------------> M2-1A (+) 
R+   <-+      (R+ = T+)
T-    <---+---------------> M2-1B  (-)
R-    <---+  (R- = T-)

I also have the resistor between (+) and (-) lines (which is the right one!) but i've tried also without and i get the same thing.

USB plugged in, i get the serial setup in /dev/ttyUSB0 and it works perfectly with libmodbus and at any speed, except the systematic Tx echo in Rx that i need to cancel by software i.e. swallow back all the transmitted bytes in a preliminary read (at the term of the _modbus_rtu_send) before attempting reading the modbus response.

I understand from what you explained that my USB-RS485 hardware should theoretically  do that automatically (what is claimed the 'automatic direction control'), for instance by disabling probing the Rx signal during transmission so as to avoid that the transmitted bytes get piled back in the Rx buffer.
Is my understanding correct?
if yes, do you reckon it is a common feature of RS485 adapters to do that ? 

if you confirm, then i guess i will just have to throw my adapter in the bin and get hold of another one that definitly supports it...

thanks again for your advice!

oliviercommelarbre, 2017-06-12:
not sure you are any interested, but just to let everyone know i've finally solved my pb by changing the adapter. 
This is what i bought: https://www.amazon.it/gp/product/B01N3LM0PU/ref=od_aui_detailpages00?ie=UTF8&psc=1

 a lot cheaper, awfully simple to wire (3 wires only GND/A/B), plug-and-play install (nothing to do except plug it in), and without that spurious echoing behaviour. just what one needs to speak modbus-rtu on a RPi via USB.

closed 2017-06-12 by oliviercommelarbre

stephane, 2017-06-14:
Thank you for the feedback